armbian/build #7507: How to install 3d acceleration on rk3588

eternal-Eason, 2024-11-23:
Hello, I would like to ask, after compiling the desktop version of Ubuntu 22.04 for Rock5B (RK3588) on Armbian, I don't have 3D acceleration, that is, Panfrost. How can I install and enable it? Thank you.

EvilOlaf, 2024-11-23:
Tried building with mesa-vpu extension?

eternal-Eason, 2024-11-23:
> Tried building with mesa-vpu extension?

@EvilOlaf  
Add to compile like this?

DESKTOP_APPGROUPS_SELECTED='browsers mesa-vpu desktop_tools internet remote_desktop' 

I added and compiled again to make sure panfork was not installed successfully

igorpecovnik, 2024-11-23:
ENABLE_EXTENSIONS (comma-separated list )
https://docs.armbian.com/Developer-Guide_Build-Switches/#advanced

eternal-Eason, 2024-11-23:
The RK3588 can now use 3D acceleration normally. Thank you for your support.

closed 2024-11-23 by igorpecovnik