armbian/build #7525: [Feature Request]: provide LTS kernel packages

LunarLambda, 2024-11-29:
### Which feature would you like to have?

I am running ZFS on a NanoPC CM3588 NAS board.

As I understand, I have two options for which kernel to run:

`linux-image-vendor-rk35xx` (6.1.75) and `linux-image-current-rockchip-rk3588` (6.12)

I cannot use the 6.12 kernel because ZFS only supports up to 6.10 currently (6.11 once ZFS 2.3.0 releases)

I'm not too happy about being tied to the vendor kernel, as it seems to have odd integration issues such as DKMS not recompiling modules automatically after kernel upgrades, instead I have to do so manually. The Armbian Discord gave me the advice of trying the mainline kernel package, however as stated I cannot do so.

Providing LTS kernel packages (currently 6.6) would hopefully solve both of these issues for me.

There's also `linux-image-6.10.6+bpo-arm64` (as well as 6.9 and 6.11) in the package repositories, however I'm not sure this can be used with my board.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

amazingfate, 2024-11-29:
6.12 should be the first lts supporting rk3588.

igorpecovnik, 2024-11-29:
> I'm not too happy 

We share frustration with you. 

> Providing LTS kernel packages (currently 6.6) would hopefully solve both of these issues for me.

That is not possible to provide. You need to use demo kernel. We have mainline kernels with experimental support, but quality is problematic from different angles. Closing as this is not possible to deliver before Q1 - Q2/2025.

closed 2024-11-29 by igorpecovnik