armbian/build #7541: [Feature Request]: Radxa Zero pin mapping

way5, 2024-12-05:
### Which feature would you like to have?

The [libgpiod](https://github.com/brgl/libgpiod) functionality is also based on correct pin names, ex:

```bash
gpioinfo
gpiochip0 - 85 lines:
        line   0:     "PIN_27"       unused   input  active-high
        line   1:     "PIN_28"       unused   input  active-high
        line   2:      "PIN_7"       unused   input  active-high
        line   3:     "PIN_11"       unused   input  active-high
        line   4:     "PIN_13"       unused   input  active-high
        line   5:     "PIN_15"       unused   input  active-high
        line   6:     "PIN_18"       unused   input  active-high
        line   7:     "PIN_40"       unused   input  active-high
        line   8:      unnamed       unused   input  active-high
        line   9:      unnamed       unused   input  active-high
...
```

Accordingly to the sources: [https://github.com/torvalds/.../meson-g12a-gpio.h](https://github.com/torvalds/linux/blob/master/include/dt-bindings/gpio/meson-g12a-gpio.h) these pins are mapped in sources, however on my RZero the result given by the above utility look like this:

```bash
gpioinfo
gpiochip0 - 85 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
...
```

Radxa developers say the pin names must be defined in device trees. 

Where I could set these names? 

Thank you!

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

way5, 2024-12-06:
Related Q&A in `libgpiod` project: https://github.com/brgl/libgpiod/issues/113

EvilOlaf, 2024-12-07:
Here is an example how to do that: https://github.com/armbian/build/pull/7403

way5, 2024-12-09:
Someone please confirm is the path I am putting [the patch](https://github.com/armbian/build/commit/b15661add51607593002775b0e100bf7f7937a13) file is correct (`meson64 kernel-6.6`) so I could create pull request. 
Thanks.

SteeManMI, 2024-12-09:
You would generally need to patch the meson64-6.6 (current) tree as well as the meson64-6.12 (edge) tree.  Otherwise when we move 'current' to 6.12 your patch will be lost. (Unless of course the patch is already included in upstream mainline after 6.6)

way5, 2024-12-09:
@SteeManMI, alright. [Done](https://github.com/armbian/build/commit/1206f426470994e135ff96863767e0a7b36346d8).

way5, 2024-12-11:
I did tests on the real HW using target kernel versions, Ubuntu and Debian. All works fine. 
So the [pull request](https://github.com/armbian/build/pull/7563) is now created please review it.

way5, 2024-12-11:
Issue fixed