armbian/build #7544: [Bug]: Radxa Zero device tree overlays

way5, 2024-12-06:
### What happened?

Preface: 
A couple of problems which were inherited from old Kernels and are persisting since "Bullseye" as far as I can recall.

### Issue # 1 - PWM:

Overlay `meson-g12a-radxa-zero-pwmao-a-on-gpioao-11` enabled but `/sys/class/pwm/pwmchip2/pwm0` doesn't created. This is pin 40 (gpiochip1, line 11). 
In comparison overlay `meson-g12a-radxa-zero-pwm-c-on-gpiox-8` do create `/sys/class/pwm/pwmchip0/pwm0` correctly and PWM works well. 
I suspect the problem is in the chip ID number `pwmchip2`, maybe it must be `1`...? 

### Issue # 2 - I2C:

- meson-g12a-radxa-zero-i2c-ao-m0-gpioao-2-gpioao-3
- meson-g12a-radxa-zero-i2c-ee-m1-gpioh-6-gpioh-7
- meson-g12a-radxa-zero-i2c-ee-m1-gpiox-10-gpiox-11
- meson-g12a-radxa-zero-i2c-ee-m3-gpioa-14-gpioa-15

No one of the above overlays does a thing. By default, there are bus 3 and bus 5 available despite the state of overlays. This can be easily confirmed by using (overlays disabled):

```bash
radxa-zero:~:# ls /dev | grep i2c     
i2c-3
i2c-5
radxa-zero:i2c-dev:# /sys/class/i2c-dev/i2c-
i2c-3@  i2c-5@
radxa-zero:~:# i2cdetect -r -y 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
radxa-zero:~:# i2cdetect -r -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

```

Although overlays remain disabled there are 2 buses active: 

- Bus 3 - (Pin # 3 - I2C_EE_M3_SDA and Pin # 5 - I2C_EE_M3_SCL) = meson-g12a-radxa-zero-i2c-ee-m3-gpioa-14-gpioa-15
- Bus 5 - is not accessible by any means (`i2c5: read_byte: Access error: Input/output error`)

### Issue # 3 - SPI:

- meson-g12a-radxa-zero-spi-spidev - enabled but:

```bash
radxa-zero:c:# ls /dev | grep spi
radxa-zero:c:#
radxa-zero:c:# /sys/class/spi_                                                
spi_master/  spi_slave/
```
spi_master/  spi_slave/ - both empty

```
spi: Failed opening SPI Device. bus:/dev/spidev0.0. Error 2 No such file or directory
```

### Issue # 4 - UART:

All uart overlays disabled:

```bash
radxa-zero:~:# ls /dev | grep ttyAML
ttyAML0
radxa-zero:~:#
```

- meson-g12a-radxa-zero-uart-ao-a-on-gpioao-0-gpioao-1   - has no effect
- meson-g12a-radxa-zero-uart-ao-b-on-gpioao-2-gpioao-3.  - has no effect
- meson-g12a-radxa-zero-uart-ao-b-on-gpioao-8-gpioao-9   - has no effect
- meson-g12a-radxa-zero-uart-ee-c                                            - has no effect

How to test with 2 ssh sessions:

```bash
mraa-uart dev 0 baud 1500000 recv 1000
```

```bash
mraa-uart dev 0 baud 1500000 send test
```

### Issue # 5: 

I think this is more `armbian-config` issue, however it is related. Let me know if you want me to create a `bug report` in config repository.

```bash
/boot/armbianEnv.txt: line 8: meson-g12a-radxa-zero-spi-spidev: command not found
```

How to reproduce:

1. Run `armbian-config` -> System -> SY017   -  Manage device tree overlays
2. Go `Back`, then this message arise.
 
The overlay name in this error is changing depends on what is the last overlay in line

```bash
radxa-zero:c:# cat /boot/armbianEnv.txt                                       
verbosity=1
console=both
overlay_prefix=meson
fdtfile=amlogic/meson-g12a-radxa-zero.dtb
rootdev=UUID=8ccc1a16-3c43-4012-a249-a3f84e8a6078
rootfstype=ext4
fdt_overlays=
overlays=meson-g12a-radxa-zero-gpio-10-led meson-g12a-radxa-zero-spi-spidev
param_spidev_spi_bus=0 
param_spidev_max_freq=10000000
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

### How to reproduce?

Enabling corresponding overlays.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

defencedog, 2024-12-08:
Excellent post

way5, 2024-12-12:
Alright, I've made some adjustments to Issue # 1 described above. 
Despite positive feedback of [mraa](https://github.com/way5/mraa) test program (`mraa/examles/c/pwm.c`) the output remains the same on both pins (18, 40), I am beginning to suspect there is something wrong with my board although GPIOs are all good, switching hi/low without issues. 
I double checked all device trees from `patches->meson-g12a-radxa-zero.dts` to `meson-g12a-gpio.h` all seems correct to me, I cannot find a contradiction or error. Register addresses look okay as well...

Going back to PWM:

> `PWM pin 18`

<details>
<summary>/boot/armbianEnv.txt </summary>

```bash
radxa-zero:c:# cat /boot/armbianEnv.txt                                                
verbosity=1
console=both
overlay_prefix=meson
fdtfile=amlogic/meson-g12a-radxa-zero.dtb
rootdev=UUID=f916b991-6ffa-4d5f-ac25-a1d9286c6419
rootfstype=ext4
overlays=meson-g12a-radxa-zero-gpio-10-led meson-g12a-radxa-zero-pwmao-a-on-gpioao-11 meson-g12a-radxa-zero-pwm-c-on-gpiox-8 meson-g12a-radxa-zero-uart-ao-a-on-gpioao-0-gpioao-1
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

</details> 

```bash
radxa-zero:c:# ./pwm                                                                     
PWM value is 0.010000
PWM value is 0.020000
PWM value is 0.030000
PWM value is 0.040000
PWM value is 0.049998
PWM value is 0.059998
PWM value is 0.069998
...

radxa-zero:~:# tail -f /var/log/syslog
2024-12-12T16:30:31.870732-06:00 radxa-zero libmraa[2797]: gpio: support for chardev interface is activated
2024-12-12T16:30:31.871115-06:00 radxa-zero libmraa[2797]: libmraa initialised for platform 'Radxa Zero' of type 40
2024-12-12T16:30:31.871395-06:00 radxa-zero libmraa[2797]: pwm_init: pwmchip0/pwm0 already exported, continuing
...

radxa-zero:~:# cat /sys/class/pwm/pwmchip0/pwm0/enable 
1
radxa-zero:~:# cat /sys/class/pwm/pwmchip0/pwm0/period 
200000

```

device enable and seems to be running but there is no effect on physical pin.

[image]

I need help here.

pyavitz, 2024-12-29:
My guess is that they aren't actually being applied at boot. The overlay_prefix is `meson` so the naming convention should be: `overlays=g12a-radxa-zero-gpio-10-led g12a-radxa-zero-spi-spidev` I believe.

One of my units sits on my desk for testing purposes and the following overlays being used.

```
meson-g12a-radxa-zero-gpio-8-led.dtbo
meson-g12a-radxa-zero-rtc-ds1307.dtbo [CUSTOM]
meson-g12a-radxa-zero-i2c-ee-m1-gpiox-10-gpiox-11.dtbo
```

```
clock: ~  $ dmesg | grep rtc
[    2.733396] meson-vrtc ff8000a8.rtc: registered as rtc1
[    2.777050] rtc-ds1307 3-0068: registered as rtc0
[    2.779066] rtc-ds1307 3-0068: setting system clock to 2024-11-15T00:23:35 UTC (1731630215)
```

```
clock: ~  $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

way5, 2025-01-12:
@pyavitz you mention the naming convention. Interesting, I never thought about that. Going to try to change the overlay names.

I am also trying to figure out how the existent devices are created (i2c for instance) if overlays having no effect in my case.

alexcaoys, 2025-08-10:
Hi @way5 ,
I was searching something related to the i2c bus on radxa zero and came across this issue. 
Just want to point out that the i2c5 is the i2c bus of hdmitx (you can find more about it if you search ddc/ddcutil). According to the schematic, hdmitx also connects to i2c3 (GPIOH_0 and GPIOH_1 pins). I guess that's why they are both enabled by default. 
Not sure if this issue is fixed or not. But I can confirm the i2c overlays is working if they are the same as [this](https://github.com/radxa/kernel/blob/linux-5.10.y-radxa-zero/arch/arm64/boot/dts/amlogic/overlay/meson-g12a-i2c-ee-m3-gpioa-14-gpioa-15.dts). I am on Armbian 25.5.2 (linux 6.12.32-current-meson64) and I am using the same syntax as this dts to enable DDC features on i2c3.