armbian/build #7548: [Bug]: bullseye: W: GPG error: https://github.armbian.com/configng stable InRelease. signatures couldn't be verified

codesnas, 2024-12-08:
### What happened?

```
[🥑] Group apt_lists_copy_from_host_to_image_and_update
  [🐳|🌱] Copying host-side apt list cache into image [ apt-get update and clean image-side ]
  [🐳|🔨]   mkdir: created directory '/armbian/.tmp/rootfs-99e5a529-d78b-4c15-88c3-a489a7e17add/var/lib/apt/lists'
  [🐳|🌱] After copying host-side cache into image [ 39 files ]
  [🐳|🌱] Updating apt lists in image [ apt-get update and clean ]
  [🐳|🔨]   Hit:1 http://deb.debian.org/debian bullseye InRelease
  [🐳|🔨]   Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
  [🐳|🔨]   Hit:3 http://deb.debian.org/debian bullseye-backports InRelease
  [🐳|🔨]   Hit:5 http://security.debian.org/ bullseye-security InRelease
  [🐳|🔨]   Get:6 https://github.armbian.com/configng stable InRelease [3,264 B]
  [🐳|🔨]   Get:4 http://netcup.armbian.com/apt bullseye InRelease [53.3 kB]
  [🐳|🔨]   Err:6 https://github.armbian.com/configng stable InRelease
  [🐳|🔨]     The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93D6889F9F0E78D5
  [🐳|🔨]   Get:7 http://netcup.armbian.com/apt bullseye/main all Packages [15.6 kB]
  [🐳|🔨]   Get:8 http://netcup.armbian.com/apt bullseye/main arm64 Packages [1,159 kB]
  [🐳|🔨]   Get:9 http://netcup.armbian.com/apt bullseye/main arm64 Contents (deb) [19.1 MB]
  [🐳|🔨]   Get:10 http://netcup.armbian.com/apt bullseye/main all Contents (deb) [38.6 kB]
  [🐳|🔨]   Get:11 http://netcup.armbian.com/apt bullseye/bullseye-utils all Packages [1,565 B]
  [🐳|🔨]   Get:12 http://netcup.armbian.com/apt bullseye/bullseye-utils arm64 Packages [12.0 kB]
  [🐳|🔨]   Get:13 http://netcup.armbian.com/apt bullseye/bullseye-utils arm64 Contents (deb) [11.3 kB]
  [🐳|🔨]   Get:14 http://netcup.armbian.com/apt bullseye/bullseye-utils all Contents (deb) [391 B]
  [🐳|🔨]   Get:15 http://netcup.armbian.com/apt bullseye/bullseye-desktop all Packages [661 B]
  [🐳|🔨]   Get:16 http://netcup.armbian.com/apt bullseye/bullseye-desktop arm64 Packages [6,212 B]
  [🐳|🔨]   Get:17 http://netcup.armbian.com/apt bullseye/bullseye-desktop all Contents (deb) [318 B]
  [🐳|🔨]   Get:18 http://netcup.armbian.com/apt bullseye/bullseye-desktop arm64 Contents (deb) [28.6 kB]
  [🐳|🔨]   Reading package lists...
  [🐳|🔨]   W: GPG error: https://github.armbian.com/configng stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93D6889F9F0E78D5
  [🐳|🔨]   E: The repository 'https://github.armbian.com/configng stable InRelease' is not signed.
  [🐳|💥] Error 100 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
      run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                    chroot_sdcard() --> lib/functions/logging/runners.sh:122
     apt_lists_copy_from_host_to_image_and_update() --> lib/functions/rootfs/apt-install.sh:67
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:56
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
   ]
  Error:  Error 100 occurred in main shell at /armbian/lib/functions/logging/runners.sh:211
      run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                    chroot_sdcard() --> lib/functions/logging/runners.sh:122
     apt_lists_copy_from_host_to_image_and_update() --> lib/functions/rootfs/apt-install.sh:67
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:56
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
  
  [🐳|💥] Cleaning up [ please wait for cleanups to finish ]
  [🐳|🌿] Unmounting recursively [ SDCARD - be patient ]
  [🐳|🌿] Unmounting recursively [ MOUNT - be patient ]
  [🐳|🌿] ANSI log file built; inspect it by running: [ less -RS output/logs/log-build-99e5a529-d78b-4c15-88c3-a489a7e17add.log.ans ]
  [🐳|🌱] SHARE_LOG=yes, uploading log [ uploading logs ]
  [🐳|🌿] Log uploaded, share URL: [ <!DOCTYPE html> <html lang=en> <head> <meta charset=utf-8> <title>Error</title> </head> <body> <pre>Cannot POST /log</pre> </body> </html> ]
  [🐳|✨] Added GHA output [ 'logs_url'='<!DOCTYPE html> <html lang=en> <head> <meta charset=utf-8> <title>Error</title> </head> <body> <pre>Cannot POST /log</pre> </body> </html>' ]
  [💥] -------------Docker run failed after 288s-------------------------- [ 🐳 failed ]
  [🌱] Cleaning up [ please wait for cleanups to finish ]
  Error: Process completed with exit code 100.
```

### How to reproduce?

```
# build bullseye
./compile.sh build BOARD=odroidn2 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no BUILD_ONLY=default COMPRESS_OUTPUTIMAGE=sha EXPERT=yes HOST=armbian KERNEL_CONFIGURE=no RELEASE=bullseye SHARE_LOG=yes
```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-12-09:
Can you try to build supported OS, bookworm, jammy, noble?

codesnas, 2024-12-11:
Thanks to the administrator, these can all be compiled normally. Is bullseye no longer supported now?

igorpecovnik, 2024-12-11:
> these can all be compiled normally.

Ticket can be closed then. 

> Is bullseye no longer supported now?

No, not anymore. For almost a year, just before it was not broken and nobody noticed. Here in the folder you can check statuses https://github.com/armbian/build/tree/main/config/distributions but I believe it should be a warning in the build log.

It might work and in theory it should be supported upstream.
https://www.debian.org/releases/bullseye/
_The Debian 11 life cycle encompasses five years: the initial three years of full Debian support, until August 14th, 2024, and two years of Long Term Support (LTS), until August 31st, 2026._

This problem is probably related to key management which has been changed some time ago, but problems are mounting up, today this, tomorrow something else, unrelated .... while our team, which is focused into hardware, can't afford, it is also stupid TBH, to maintain so many different (and outdated) build targets. Especially as there is [almost complete absence](https://github.com/sponsors/armbian) of public support.

If you really need to build this old target, you will need to spent some time and fix / workaround it. Even we could put this on our bug list, it won't be addressed anytime soon. Certainly not this year and most likely not even next. Fixing old releases has very low priority. I hope you do understand all this.

closed 2024-12-12 by codesnas

igorpecovnik, 2024-12-13:
This will fix it:
https://github.com/armbian/build/pull/7524