armbian/build #7551: [Bug]: U-boot trying to load dtb from the wrong location (and failed)

chemwolf6922, 2024-12-08:
### What happened?

I'm using a 64bit board (nanopi neo 2). The overlay dtbs are located at `/boot/dtb/allwinner/overlay/` as they should. But when booting, the following u-boot log was printed:
```Shell
Failed to load '/boot/dtb/overlay/sun50i-h5-spi-spidev.dtbo'
Failed to load '/boot/dtb/overlay/sun50i-h5-fixup.scr'
```
u-boot is trying to load overlays in the 32bit system folder.

### How to reproduce?

Flash a new system 
```
Linux nanopineo2 6.6.62-current-sunxi64 #1 SMP Sun Nov 17 14:08:59 UTC 2024 aarch64 GNU/Linux
Armbian_community 25.2.0-trunk.124 Bookworm 
```
Enable the spi-spidev by adding the following lines to `/boot/armbianEnv.txt`
```
overlays=spi-spidev
param_spidev_spi_bus=0
param_spidev_max_freq=100000000
```
Reboot. Look at the u-boot log.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chemwolf6922, 2024-12-08:
The fdtfile u-boot env is wrong. It is set to 
```
fdtfile=allwinner/sun50i-h5-nanopi-neo2.dtb
```
This will lead to a wrong fdtdir (`dtb`). Which will lead to the wrong overlay files:
```
${fdtdir}/overlay/${overlay_prefix}-${overlay_file}.dtbo
```

chemwolf6922, 2024-12-08:
> The fdtfile u-boot env is wrong. It is set to
> 
> ```
> fdtfile=allwinner/sun50i-h5-nanopi-neo2.dtb
> ```
> 
> This will lead to a wrong fdtdir (`dtb`). Which will lead to the wrong overlay files:
> 
> ```
> ${fdtdir}/overlay/${overlay_prefix}-${overlay_file}.dtbo
> ```

A temporary fix: Adding this line to /boot/armbianEnv.txt to override the wrong env. 
```
fdtfile=sun50i-h5-nanopi-neo2.dtb
```

igorpecovnik, 2024-12-08:
fix in the pr stage https://github.com/armbian/build/pull/7535

The-going, 2024-12-08:
> The fdtfile u-boot env is wrong. It is set to
> 
> ```
> fdtfile=allwinner/sun50i-h5-nanopi-neo2.dtb
> ```

A temporary solution to the problem:
Please add to armbianEnv.txt file string `fdtfile=sun50i-h5-nanopi-neo2.dtb`
without vendor's name

I'm sorry for the inconvenience.

chemwolf6922, 2024-12-10:
Thanks for the reply. I'm happy to help. Closing this one since the fix was merged.