armbian/build #7555: [Bug]: Built-in Wi-Fi is not working on the Orange Pi 5 Max

martinbone, 2024-12-08:
### What happened?

The board has the : AP6611 bluetooth / Wi-Fi module.
Bluetooth works ... but for some reason, Wi-Fi does not.

On Ubuntu Rockchip, both bluetooth and Wi-Fi are working.

### How to reproduce?

Go to settings - WI-FI. Message reads: "No Wi-Fi adapter found".

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amazingfate, 2024-12-09:
Can't reproduce, tested with image https://github.com/armbian/community/releases/download/25.2.0-trunk.124/Armbian_community_25.2.0-trunk.124_Orangepi5-max_noble_vendor_6.1.84_gnome_desktop.img.xz

yswtrue, 2025-01-11:
Same here build by
```
./compile.sh build BOARD=orangepi5-max BRANCH=vendor BUILD_DESKTOP=yes BUILD_MINIMAL=no DESKTOP_APPGROUPS_SELECTED='browsers chat desktop_tools editors internet multimedia office programming remote_desktop' DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base KERNEL_CONFIGURE=no RELEASE=bookworm
```
and the gui is very slow, no usb 3.0 support

igorpecovnik, 2025-01-11:
> and the gui is very slow, no usb 3.0 support

Consult with manufacturer and [community](https://forum.armbian.com/forum/97-community-maintained-staging/) . This is not Armbian firmware bug.

closed 2025-01-11 by igorpecovnik