armbian/build #7590: [Feature Request]: Eenable HDMI options for RK3399

ihipop, 2024-12-17:
### Which feature would you like to have?

CAN we revoke this https://github.com/armbian/build/pull/2016 , I really want HDMI output

LibrELEC uses mainline kernel and mainline bootloader too

they could make HDMI works

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

amazingfate, 2024-12-17:
#2016 just disbale hdmi support in u-boot. HDMI is still supported in kernel. I have rockpi 4b which has fine HDMI support. Feel free to add pr to solve issues of your board.

ihipop, 2024-12-17:
> #2016 just disbale hdmi support in u-boot. HDMI is still supported in kernel. I have rockpi 4b which has fine HDMI support. Feel free to add pr to solve issues of your board.

I can't get HDMI output in RK3399 board Nano PC T4, what I have tried https://dl.armbian.com/nanopct4/Noble_current_gnome-kisak  https://dl.armbian.com/nanopct4/Bookworm_current_minimal
the SSH is working 

I can only get a black screen on HDMI

amazingfate, 2024-12-17:
NanoPC T4 is a board supported by community, and I don't have this board, so I can't help you. You can post your issue at forum for further help: https://forum.armbian.com/tags/nanopct4/

ihipop, 2024-12-17:
Moved to https://forum.armbian.com/topic/48372-hdmi-output-for-nanopc-t4-is-broken-since-linux-kernel-6x/

closed 2024-12-17 by ihipop