armbian/build #7612: [Bug]: rpi4b build fails with BOOTFS_TYPE="fat"

MSzturc, 2024-12-24:
### What happened?

  When building with BOOTFS_TYPE="fat" the rpi4b failed with the following error message:

[🐳|🌱] Populating /boot/firmware directory [ bcm2711 ]
  [🐳|🔨]   cp: cannot stat '/boot/vmlinuz-6.6.45-current-bcm2711': No such file or directory
  [🐳|💥] Error 1 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
      run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                    chroot_sdcard() --> lib/functions/logging/runners.sh:122
     post_family_tweaks__populate_boot_firmware_directory() --> config/sources/families/bcm2711.conf:184
               post_family_tweaks() --> <extension_magic>:125
            call_extension_method() --> lib/functions/general/extensions.sh:57
     install_distribution_agnostic() --> lib/functions/rootfs/distro-agnostic.sh:393
                  do_with_logging() --> lib/functions/logging/section-logging.sh:81
           build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:28
     full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
            do_with_default_build() --> lib/functions/main/default-build.sh:42
           cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
          armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                   cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                             main() --> compile.sh:50
   ]
  Error:  Error 1 occurred in main shell at /armbian/lib/functions/logging/runners.sh:211

### How to reproduce?

Build with following configuration:

ARMBIAN_REPOSITORY="armbian/build"
#ARMBIAN_BRANCH="v24.08"
ARMBIAN_BRANCH="main"
KERNEL_CONFIGURE=prebuilt
CLEAN_LEVEL=""  
INSTALL_HEADERS="no"
DOWNLOAD_MIRROR="" 
CARD_DEVICE=""    
BETA="no"             
BRANCH="current"           
BOOTFS_TYPE="fat"           
COMPRESS_OUTPUTIMAGE="img"  
VENDOR="Armbian"
BOARD=rpi4b
RELEASE="bookworm"
KERNEL_CONFIGURE=prebuilt
BUILD_DESKTOP=no
BUILD_MINIMAL=yes

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-12-24:
lacking full logs

MSzturc, 2024-12-24:
added full log

EvilOlaf, 2024-12-24:
Interesting. Since building w/o fat bootfs works just fine may I ask why you need this?
Also /boot/firmware is fat32 by default from what I can see in the build logs.

MSzturc, 2024-12-24:
The boot partition allows users of our operating system to configure settings before the initial startup. This workflow is currently implemented on Banana Pi and Orange Pi devices. For Raspberry Pi-based boards, we currently rely on Raspberry Pi OS Lite. However, our goal is to streamline and unify our build pipeline by transitioning entirely to Armbian.

EvilOlaf, 2024-12-25:
I see. Well since there are literally hundreds of issues spread across Github, Jira and forums needing attention it is impossible to tell when or if this gets addressed. If this is business critical for your we offer commercial support too.

MSzturc, 2024-12-26:
It’s not a critical issue for us, as we have a reliable and stable build pipeline built on Raspberry Pi OS Lite.

rpardini, 2024-12-26:
> This workflow is currently implemented on Banana Pi and Orange Pi devices

True, and also on all other u-boot based systems in some way or another.

Currently, our rpi build _isn't_ u-boot based; instead, it uses the rpi-proprietary bootloader which loads config.txt (and kernel/initrd/dt/cmdline.txt) directly from a FAT32 "firmware" partition.

closed 2024-12-26 by rpardini