armbian/build #7619: [Feature Request]: Create a new build using edge kernel for FriendlyElec CM3588 NAS

eabase, 2024-12-27:
### Which feature would you like to have?

See comment it last build:

* https://github.com/armbian/build/pull/6355

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2024-12-28:
What prevents you from building yourself?
https://docs.armbian.com/User-Guide_FAQ/#why-is-there-no-image-for-board-with-bookwormjammynobletrixie-and-minimalclignomekdexfce-with-vendorlegacycurrentedge-kernel

eabase, 2024-12-28:
Because (1) I did not find those pages, and (b) the pages you linked are convoluted and and unclear, with no build instructions.

* https://docs.armbian.com/User-Guide_FAQ/#how-can-i-compile-my-own-kernel

> Normally on Debian or Ubuntu you would do something like: 
> `sudo apt-get build-dep linux linux-image-$(uname -r)`
> However Armbian’s way of building kernel images is slightly different than the standard distribution method. The best way is to follow the procedures in the [Developer Guide](https://docs.armbian.com/Developer-Guide_Build-Preparation/).

EvilOlaf, 2024-12-28:
> short list of compile options.

Because there isn't much more to it. Clone repo, run `./compile.sh`. That's it.
The build frameworks allows building images in various combinations of kernel, userspace, desktop environment and so on very easy.

Anyway there is also a video linked on the same page showing the process.

closed 2025-01-19 by igorpecovnik