armbian/build #7641: [Bug]: Nanopineocore2 build not booting

Keksstar, 2025-01-02:
### What happened?

Downloaded Armbian for the Nano Pi Neo Core 2 (Build Date: Nov 25, 2024) from the official website: https://www.armbian.com/nanopi-neo-core-2-lts/
Both versions, Ubuntu or Debian won´t boot on the board itself.

From the FriendlyElec Wiki I´ve downloaded the Debian Bookworm image and the board boots up instantly: https://wiki.friendlyelec.com/wiki/index.php/NanoPi_NEO_Core (https://drive.google.com/drive/folders/1rcF2-Qrw9o_OACNV6cAj_Vzv-2BYDGQz)

At some point Armbian must have loosen the ability to work on this board.
Sadly I´m no programmer and can´t deep-dive into this.

### How to reproduce?

Download the image from the Armbian website (Build Date: Nov 25, 2024) ;-)

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-01-03:
HI,

this issue tracker is exclusively for problems with the build framework. _building_ Armbian to say.
Issues when running Armbian shall be reported in forum as suggested by the issue template and the bug wizzard on our website.

How to debug boot issues and get proper logs for sharing: https://debug.armbian.de

closed 2025-01-03 by EvilOlaf

igorpecovnik, 2025-01-03:
> I have other options

Not build-able. We provide old images in archives too.

igorpecovnik, 2025-01-03:
> I assume it´s framework stuff, as it worked several versions ago with this board....

Kernel support is fragile and constantly breaks boards. Most of bugs are made by kernel community as a whole, we fix what we can. FYI