armbian/build #7656: [Feature Request]: support for the odroid-M2 possible ?

mxcxpx, 2025-01-04:
### Which feature would you like to have?

iam missing the Odroid-M2 in the list.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2025-01-04:
Hi,

Armbian will not consider this request and maintenance without funding.
However anyone from the community is free to donate their time adding support for new hardware. 

On a personal note:  Support should be possible reasonably easy since I saw stuff for the Radxa ROCK 5C, which uses the same RK3588S2 SoC, being pushed to both mainline U-Boot and Linux. I suggest anyone who wants to tackle this to check upstream patch mailing lists.

rpardini, 2025-01-04:
I think Kwiboo sent ODROID-M2 patches upstream...
... kernel: https://github.com/torvalds/linux/commits/master/arch/arm64/boot/dts/rockchip/rk3588s-odroid-m2.dts
... u-boot: https://github.com/u-boot/u-boot/commit/bcc7eefa4c6cce0c719296dcdc40a353160633fc

thus should be trivial to add using 6.13-rcX and 2025.01-rcX

igorpecovnik, 2025-01-04:
I wish we would have time for endless stream of hardware ...

https://docs.armbian.com/Process_Contribute/#adding-a-new-board

EvilOlaf, 2025-01-05:
> I think Kwiboo sent ODROID-M2 patches upstream..

I did not even check what was upstream already, just noticed there are patches in the queue :grinning:

mlegenovic, 2025-02-13:
Create `config/boards/odroidm2.csc` file with this content:

```
# Rockchip RK3588s SoC octa core 4-16GB SoC eMMC USB3 NvME
BOARD_NAME="Odroid M2"
BOARDFAMILY="rockchip64"
BOARD_MAINTAINER="mlegenovic"
BOOT_SOC="rk3588"
KERNEL_TARGET="edge"
BOOT_FDT_FILE="rockchip/rk3588s-odroid-m2.dtb"
BOOT_SCENARIO="binman"
IMAGE_PARTITION_TABLE="gpt"
FULL_DESKTOP="no"
BOOT_LOGO="yes"

BOOTBRANCH_BOARD="tag:v2025.04-rc2"
BOOTBRANCH="${BOOTBRANCH_BOARD}"
BOOTPATCHDIR="v2025.04-rc2"

BOOTCONFIG="odroid-m2-rk3588s_defconfig"
BOOTDIR="u-boot-${BOARD}" # do not share u-boot directory

OVERLAY_PREFIX='rockchip-rk3588'

function post_family_config__uboot_config() {
	display_alert "$BOARD" "u-boot ${BOOTBRANCH_BOARD} overrides" "info"
	BOOTDELAY=1 # Wait for UART interrupt to enter UMS/RockUSB mode etc
	UBOOT_TARGET_MAP="BL31=${RKBIN_DIR}/$BL31_BLOB ROCKCHIP_TPL=${RKBIN_DIR}/${DDR_BLOB};;u-boot-rockchip.bin"
	unset uboot_custom_postprocess write_uboot_platform write_uboot_platform_mtd # disable stuff from rockchip64_common; we're using binman here which does all the work already

	# Just use the binman-provided u-boot-rockchip.bin, which is ready-to-go
	function write_uboot_platform() {
		dd if=${1}/u-boot-rockchip.bin of=${2} bs=32k seek=1 conv=fsync
	}
}

# "rockchip-common: boot SD card first, then NVMe, then SATA, then USB, then mmc"
# On odroidm2, mmc0 is the eMMC, mmc1 is the SD card slot
function pre_config_uboot_target__odroidm2_patch_rockchip_common_boot_order() {
	declare -a rockchip_uboot_targets=("mmc1" "nvme" "mmc0" "usb" "pxe" "dhcp") # for future make-this-generic delight
	display_alert "u-boot for ${BOARD}/${BRANCH}" "u-boot: adjust boot order to '${rockchip_uboot_targets[*]}'" "info"
	sed -i -e "s/#define BOOT_TARGETS.*/#define BOOT_TARGETS \"${rockchip_uboot_targets[*]}\"/" include/configs/rockchip-common.h
	regular_git diff -u include/configs/rockchip-common.h || true
}

# add a network rule to rename default name
function post_family_tweaks__odroidm2_rename_gmac_eth0() {
	display_alert "Creating network rename rule for Odroid M2"
	mkdir -p "${SDCARD}"/etc/udev/rules.d/
	cat <<- EOF > "${SDCARD}"/etc/udev/rules.d/70-rename-lan.rules
		SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", KERNEL=="end*", NAME="eth0"
	EOF

}
```

Build with `./compile.sh build BOARD=odroidm2 BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=noble` and have fun with experimenting.

EvilOlaf, 2025-02-13:
Feel free to send PR. CSC targets can be added anytime.

mlegenovic, 2025-02-13:
@EvilOlaf It's still too early for a PR, because this is only the first version that was intended to get the whole thing up and running. I haven't tested it intensively yet, and there are also some things that can be improved with patches, e.g. audio via HDMI. One user has also reported that the network is unstable (he is playing with archlinux distribution), which I haven't had so far.

I have posted it here so that other people can also try it out and possibly discover things that can be improved/fixed before PR.

Sources:

- [HDMI audio support on the rk3588 based devices](https://lore.kernel.org/linux-rockchip/20250203171925.126309-1-detlev.casanova@collabora.com/)
- [Support for es8316 analog audio](https://github.com/mth/u-boot-odroid-m2/tree/current/patches)
- [Fix for ethernet work with stock U-Boot](https://github.com/tobetter/linux/commit/3b3b1bcbce41e5800daabd2277c682f56d80e8d4)

closed 2025-03-05 by EvilOlaf