armbian/build #7708: [Bug]: building any board occurre an internel hashtable error

ZYQsteve, 2025-01-16:
### What happened?

 [Bug]: building any board occurre an internel hashtable error on step nemed copy files to [/boot (mount /boot)] 
`Copying files to [ /boot (MOUNT /boot) ]
[🔨]   Internal hashtable error: illegal key supplied!
[🔨]   rsync error: errors with program diagnostics (code 13) at hashtable.c(88) [generator=3.2.7]
[💥] Error 13 occurred in main shell [ at /home/steve/test/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        run_host_command_logged() --> lib/functions/logging/runners.sh:193
   create_image_from_sdcard_rootfs() --> lib/functions/image/rootfs-to-image.sh:80
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:93
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
[🌿] Unmounting recursively [ SDCARD - be patient ]
[🌿] Unmounting recursively [ MOUNT - be patient ]
[🚸] Freeing loop [ trap_handler_cleanup_rootfs_and_image /dev/loop16 ]
`

### How to reproduce?

run compile.sh to compile any board

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-16:
> run compile.sh to compile any board

Please provide exact commands to reproduce.

dellagd, 2025-01-16:
Same issue.

Ran command: `./compile.sh build BOARD=rock-5c BRANCH=vendor BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=jammy`

Possibly related: https://github.com/NixOS/nixpkgs/pull/373845, https://github.com/RsyncProject/rsync/issues/702

ZYQsteve, 2025-01-16:
./compile.sh build BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=no KERNEL_CONFIGURE=no RELEASE=jammy HTTPS_PROXY="$https_proxy" HTTP_PROXY="$http_proxy"

ZYQsteve, 2025-01-16:
solved by upgrade rsync to v3.4.1

closed 2025-01-16 by ZYQsteve

ct991218620, 2025-01-16:
我也遇到了同样的问题，但是用同样的方式之前还可以编译成功的，是不是服务出现了什么问题。

dellagd, 2025-01-16:
@ZYQsteve Where exactly did you upgrade rsync? My compile command is running with Docker as a backend so it's not my host rsync that I have to update I believe.

dellagd, 2025-01-16:
For anyone else hitting this running through docker, I was able to patch around this by doing an install from source of rsync around this point: https://github.com/armbian/build/blob/f4457a3df56fccd5701f259336e4aa395b13305f/lib/functions/host/docker.sh#L283

To get this working without a patch, I _think_ an updated package needs to be pushed to the Jammy ubuntu APT repository. No idea what the process for that looks like, or if the rsync maintainers will be doing that.

ZYQsteve, 2025-01-16:
> 我也遇到了同样的问题，但是用同样的方式之前还可以编译成功的，是不是服务出现了什么问题。

我没使用dockers编译，直接使用真机编译，更新是通过下载rsync源码编译更新的

ZYQsteve, 2025-01-16:
> [@ZYQsteve](https://github.com/ZYQsteve) Where exactly did you upgrade rsync? My compile command is running with Docker as a backend so it's not my host rsync that I have to update I believe.

I upgrade the rsync by their official source code on GitHub, without using Docker