armbian/build #7710: [Bug]: Armbian skips rebuild updated U-boot and keeps cached

bmx666, 2025-01-16:
### What happened?

I'm using custom board config that has a similar as rock-5b.conf U-boot post config

```
# Mainline u-boot or Kwiboo's tree
function post_family_config_branch_edge__rock-5b_use_mainline_uboot() {
        display_alert "$BOARD" "mainline (next branch) u-boot overrides for $BOARD / $BRANCH" "info"

        declare -g BOOTCONFIG="rock5b-rk3588_defconfig"                       # override the default for the board/family
        declare -g BOOTDELAY=1                                                # Wait for UART interrupt to enter UMS/RockUSB mode etc
        declare -g BOOTSOURCE="https://github.com/Kwiboo/u-boot-rockchip.git" # We ❤️ Kwiboo's tree
        declare -g BOOTBRANCH="branch:rk3xxx-2024.04"                         # commit:31522fe7b3c7733313e1c5eb4e340487f6000196 as of 2024-04-01
        declare -g BOOTPATCHDIR="v2024.04-rock5b-radxa"                       # empty; defconfig changes are done in hook below
        declare -g BOOTDIR="u-boot-${BOARD}"                                  # do not share u-boot directory
        declare -g UBOOT_TARGET_MAP="BL31=${RKBIN_DIR}/${BL31_BLOB} ROCKCHIP_TPL=${RKBIN_DIR}/${DDR_BLOB};;u-boot-rockchip.bin u-boot-rockchip-spi.bin"
        unset uboot_custom_postprocess write_uboot_platform write_uboot_platform_mtd # disable stuff from rockchip64_common; we're using binman here which does all the work already

        # Just use the binman-provided u-boot-rockchip.bin, which is ready-to-go
        function write_uboot_platform() {
                dd "if=$1/u-boot-rockchip.bin" "of=$2" bs=32k seek=1 conv=notrunc status=none
        }

        function write_uboot_platform_mtd() {
                flashcp -v -p "$1/u-boot-rockchip-spi.bin" /dev/mtd0
        }
}
```

If I push new commits into branch - Armbian just skips U-boot build and use cached version.

```
[🐳|🌱] Using cached pip packages for Python tools [ a61a78b41c898744fbc2d1d13d6a127df20406336cbe64af66bf1e075570b93e ]
[🐳|🔨]   Summary: Aggregation: debootstrap: 5; rootfs: 45; image: 3; desktop: 0; apt-sources: 0; hash: 4a7eeaed1446ffd6
[🐳|🌱] artifact [ uboot :: uboot() ]
[🐳|🌱] Using cached [ GIT_INFO_UBOOT ]
[🐳|🌱] Reversioning package [ re-version 'uboot-customer-abc-vendor(deb)::2024.10-rc6-Sabb5-P0000-H272d-Vc176-Bb703-R448a' to '25.02.0-trunk' ]
```

### How to reproduce?

Push new commits into `BOOTBRANCH` and recompile image. Result - old U-boot in output image.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

bmx666, 2025-01-16:
Could you provide information how to FORCE rebuild U-boot ?

bmx666, 2025-01-16:
and the worse part, if you for example update `BOOTDELAY` from 1 to 2, it will build a new U-boot, but if you return back `BOOTDELAY` from 2 to 1, Armbian will use OLD CACHED U-boot. Cache is good, but no in that case, a lot of side-effects.

mr-toolcraft, 2025-01-17:
`ARTIFACT_IGNORE_CACHE=yes` will force rebuild.
So for some reason the config seems not to be considered when hashing? @rpardini

bmx666, 2025-01-20:
@mr-toolcraft , thanks a lot! `ARTIFACT_IGNORE_CACHE=yes` that's what I'm looking for.

For kernel, if there is a new commit in a branch, Armbian rebuilds the kernel. But for U-boot no reaction, even steps that should fetch the branch from the repository are ignored.

rpardini, 2025-01-25:
Untrue. Both kernel and u-boot 'react' to changes in the upstream git repo (`KERNELSOURCE` and `BOOTSOURCE`).

We do cache (in cache/memoize if I'm not mistaken) the repo's SHA1 for a certain amount of time -- otherwise the build system would be hammering the upstream git server, which is considered bad practice, especially for kernel.org.

For kernel, one can control this cache's TTL with `KERNEL_GIT_CACHE_TTL=120` (cache for 2 minutes).

I don't think this is implemented for u-boot, and thus the cache defaults to 3600 seconds.

A Pull request implementing `UBOOT_GIT_CACHE_TTL` (via `memoize_cache_ttl` var to `run_memoized()` just like is done in `obtain_kernel_git_info_and_makefile()` would be welcome.