armbian/build #7715: [Bug]: Doesn't work because git repositories are missing

easy-and-simple, 2025-01-17:
### What happened?

fatal: repository 'https://github.com/morrownr/8821cu-20210118/' not found

fatal: repository 'https://github.com/150balbes/wifi/' not found

I try to build CLI image for ROCK 3A with kernel options
Tried and debian and ubuntu same thing

### How to reproduce?

compile.sh EXPERTS='yes' NO_APT_CACHE='yes'
options have no matter

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-01-18:
no logs - no issue.

amazingfate, 2025-01-18:
Your code is out of date, the url has been fixed over a year: https://github.com/armbian/build/commit/669d00a6ab85e74e31938f7433a0b4edcae81fa2

easy-and-simple, 2025-03-16:
> Your code is out of date, the url has been fixed over a year: [669d00a](https://github.com/armbian/build/commit/669d00a6ab85e74e31938f7433a0b4edcae81fa2)

Thank you. I set git to use main origin, and now it works but my goal to build it by myself is to enable RKNPU driver in kernel because pre-built Armbian images for Rock 3A board doesn't have RKNPU driver. Making Armbian build to work doesn't solve my problem because I can't find option to enable RKNPU in kernel.
Does someone know why RKNPU is not enabled in pre-built Armbian images for Rock 3A boards, and why there is no option in kernel settings for this board in Armbian buid?
Thank you in advance!

closed 2025-03-22 by easy-and-simple

igorpecovnik, 2025-03-22:
> and why there is no option in kernel settings for this board in Armbian buid?

https://docs.armbian.com/User-Guide_FAQ/#why-does-hardware-feature-xy-work-in-old-kernel-but-not-in-more-recent-one

Missing feature in kernel support is not a build framework problem. AFAIK this feature work only in vendor (6.1.y) kernel, while in modern mainline based kernel 6.12 probably not yet.

> Does someone know

Try also here https://forum.armbian.com - check similar boards forums, not just copy / pasting question into new topic.