armbian/build #7731: [Bug]: remove directories from sha files

leggewie, 2025-01-21:
### What happened?

```
$ cat Armbian_23.8.1_Bananapim2plus_jammy_current_6.1.47_minimal.img.xz.sha
59a2f556f497bbe6a99f71a9a35736beadacaff495ab67fef198b799ba0b53d6 */armbian/.tmp/image-679d4d6c-dbbb-4915-ba5c-0f466f3924b9/Armbian_23.8.1_Bananapim2plus_jammy_current_6.1.47_minimal.img.xz
```

I think it would be better to remove the subdirectories from the publish sha files, either by switching first into the subdirectory before computing the sha256sum or after the creation via ```sed 's!\*/armbian/\.tmp/!!'``` or something similar.

Am I correct to think the creation of the sha files happens via [lib/functions/general/hash-files.sh](../tree/main/lib/functions/general/hash-files.sh)?

### How to reproduce?

download and image and try to verify via sha256sum from the CLI

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-21:
>  Armbian_23.8.1_

Can you check latest images first.

leggewie, 2025-02-09:
> > Armbian_23.8.1_
> 
> Can you check latest images first.

That's the latest release for the board in question.  In any case, it's the same for other boards, too.

```
+--[ https://dl.armbian.com/bananapim7/Noble_current_gnome-kisak.sha ]
|
| 384c9c14cd85332eb8b775c7ef12a03ff2ba9dab417a0e2e46085b893e52bd60 */armbian/.tmp/image-no-uuidgen-yet-14101-7914/Armbian_24.11.1_Bananapim7_noble_current_6.12.0-rc7_gnome-kisak_desktop.i
mg.xz
|
+--
```

igorpecovnik, 2025-02-09:
I see, yes, its a bug. Thanks.

igorpecovnik, 2025-02-09:
This works now, perhaps it can be solved in more elegant way?

leggewie, 2025-02-10:
Thank you. Well done!

Since you're asking.  There's more than 1 way to skin the cat, of course.  I am not so proficient with ```awk``` or the build code of armbian, for what it's worth.  But here are some alternative ways of doing it.

- cd into the directory before running ```sha256sum``` to avoid having to strip the information at all
- I spotted a number of uses of ```cut -d -f``` around hash-files.sh, this could be used here as well for consistency
- generic ```sed```

igorpecovnik, 2025-02-10:
I'll just go with this. Tnx, lets close this down.

closed 2025-02-10 by igorpecovnik