armbian/build #7744: [Bug]: rsync error: errors with program diagnostics (code 13) at hashtable.c(88) [generator=3.2.7]

Lemon1151, 2025-01-25:
### What happened?

🐳|🌿] Actual rootfs size [ 1658MiB ]
[🥑] Group prepare_partitions
  [🐳|🌱] Preparing image file for rootfs [ tpm312 bookworm ]
  [🐳|🌱] e2fsprogs version [ 1.46.5 does not support orphan_file ]
  [🐳|🌱] Current rootfs size [ 1658 MiB ]
  [🐳|🌱] Creating blank image for rootfs [ truncate: 2180 MiB ]
  [🐳|🌱] Creating partitions [ root: ext4 ]
  [🐳|🔨]   Checking that no-one is using this disk right now ... OK
  [🐳|🔨]   
  [🐳|🔨]   Disk /armbian/.tmp/rootfs-3ce8ce4d-dda5-4a3f-895c-12f023ecbb76.raw: 2.13 GiB, 2285895680 bytes, 4464640 sectors
  [🐳|🔨]   Units: sectors of 1 * 512 = 512 bytes
  [🐳|🔨]   Sector size (logical/physical): 512 bytes / 512 bytes
  [🐳|🔨]   I/O size (minimum/optimal): 512 bytes / 512 bytes
  [🐳|🔨]   
  [🐳|🔨]   >>> Script header accepted.
  [🐳|🔨]   >>> Created a new DOS disklabel with disk identifier 0xbfc796dc.
  [🐳|🔨]   /armbian/.tmp/rootfs-3ce8ce4d-dda5-4a3f-895c-12f023ecbb76.raw1: Created a new partition 1 of type 'Linux' and of size 2.1 GiB.
  [🐳|🔨]   /armbian/.tmp/rootfs-3ce8ce4d-dda5-4a3f-895c-12f023ecbb76.raw2: Done.
  [🐳|🔨]   
  [🐳|🔨]   New situation:
  [🐳|🔨]   Disklabel type: dos
  [🐳|🔨]   Disk identifier: 0xbfc796dc
  [🐳|🔨]   
  [🐳|🔨]   Device                                                         Boot Start     End Sectors  Size Id Type
  [🐳|🔨]   /armbian/.tmp/rootfs-3ce8ce4d-dda5-4a3f-895c-12f023ecbb76.raw1      32768 4464639 4431872  2.1G 83 Linux
  [🐳|🔨]   
  [🐳|🔨]   The partition table has been altered.
  [🐳|🔨]   Syncing disks.
  [🐳|🌿] Allocated loop device [ LOOP=/dev/loop0 ]
  [🐳|🌿] Creating device node [ /dev/loop0p1 ]
  [🐳|🌱] Device node created OK [ /dev/loop0p1 ]
  [🐳|🌿] Creating rootfs [ ext4 on /dev/loop0p1 ]
  [🐳|🔨]   tune2fs 1.46.5 (30-Dec-2021)
  [🐳|🌿] Mounting rootfs [ /dev/loop0p1 (UUID=d90c2f8c-43e0-48cf-a332-920c0130a215) ]
  [🐳|🔨]   UUID=d90c2f8c-43e0-48cf-a332-920c0130a215 / ext4 defaults,noatime,commit=120,errors=remount-ro 0 1
  [🐳|🌱] extlinux.conf exists [ removing armbianEnv.txt ]
  Section 'prepare_partitions' took 0s to execute.
[🥑] Group create_image_from_sdcard_rootfs
  [🐳|🌿] Copying files via rsync to [ / (MOUNT root) ]
  [🐳|🔨]   
  [🐳|🔨]   sent 1.69G bytes  received 757.10K bytes  177.75M bytes/sec
  [🐳|🔨]   total size is 1.69G  speedup is 1.00
  [🐳|🌿] Copying files to [ /boot (MOUNT /boot) ]
  [🐳|🔨]   Internal hashtable error: illegal key supplied!
  [🐳|🔨]   rsync error: errors with program diagnostics (code 13) at hashtable.c(88) [generator=3.2.7]

### How to reproduce?

./compile.sh

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-01-25:
Please provide full build log. Excerpts are often useless.

Lemon1151, 2025-01-25:
> Please provide full build log. Excerpts are often useless.

[0_build-armbian.txt.log](https://github.com/user-attachments/files/18544991/0_build-armbian.txt.log)

我通过自己创建的GitHub actions构建的，这是日志不知道可以不可以？

igorpecovnik, 2025-01-25:
I think rsync errors are from bug in rsync application. Just update your system to latest.

rpardini, 2025-01-25:
Yes, bug in rsync, see https://www.claudiokuenzler.com/blog/1463/rsync-security-update-introduced-regression-hashtable-error

closed 2025-01-25 by rpardini