armbian/build #7753: [Bug]: Unable to login due to hanging motd

WiseLord, 2025-01-27:
### What happened?

Recently I experience some issues with accessing `whatismyip.akamai.com` (some kind of issues from ISP). This results in hanging motd inside of get_wan_address() in https://github.com/armbian/build/blob/main/packages/bsp/common/etc/update-motd.d/10-armbian-header.

It seems passing `--connect-timeout 2` to curl is not enough to guarantee fast return when some network problems happen.

I strongly recommend adding one more curl option, i.e. `--max-time 2` - this 100% works in my case.

BTW, FYI in my own scripts for finding WAN IP I use `curl ifconfig.me` - this is shorter.

### How to reproduce?

It's actually hard to reproduce. For me curl call works when I use some VPN server, but hangs when I use my ISP network It worked two days before...)

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-27:
> It's actually hard to reproduce. 

Yeah, I can understand this. If you got few more minutes, make a PR with those changes. Most of things people recommend, especially on forums, we are unable to integrate :( 

Thank you for finding this one and ideas for solving it.

WiseLord, 2025-01-27:
Please see https://github.com/armbian/build/pull/7755

igorpecovnik, 2025-02-04:
Done, closing. Thanks again.