armbian/build #7754: [Bug]: FriendlyARM NanoPC T4 DDR unstable

msdos03, 2025-01-27:
### What happened?

FriendlyARM official system images all run fine.

"memtester" in armbian and mtest in u-boot don't report any error, but when I am using a browser or running a java virtual machine, it crashes frequently. After some tests on the u-boot, I found the issue is related to RAM.

If I use 800MHz DDR blob or modify `rk3399-nanopc-t4-u-boot.dtsi` in u-boot to include `rk3399-sdram-lpddr3-4GB-1600.dtsi` instead of `rk3399-sdram-lpddr3-samsung-4GB-1866.dtsi` , The problem is solved.

That make sense to me because FriendlyARM official system images are all using 800MHz DDR blob.

So, I suggest change the DDR blob to 800MHz one, or switch to blobless and add a patch to apply the mentioned modification.

### How to reproduce?

It is hard to reproduce on other boards because the RAM chips and board revision are diffrent. On my board, installing java and run a large program or browsing the internet for a while can reproduce the problem.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mr-toolcraft, 2025-01-27:
While this issue tracker is officially for framework problems only (_building_ Armbian to say), I'd say to leave it here for now since ideally OP could provide a PR with the mentioned changes and link this issue.

Did you try both ways blobless and with the 800 blob?

msdos03, 2025-01-27:
> While this issue tracker is officially for framework problems only (_building_ Armbian to say), I'd say to leave it here for now since ideally OP could provide a PR with the mentioned changes and link this issue.

Where should I place this issue?

> Did you try both ways blobless and with the 800 blob?

Yes, I've tried both ways, they both work fine.

closed 2025-01-30 by igorpecovnik