armbian/build #7830: [Bug]: Internal hashtable error: illegal key supplied!

the-snork, 2025-02-15:
### What happened?

See https://bugs.launchpad.net/ubuntu/+source/rsync/+bug/2095004

[🐳|🌿] Copying files via rsync to [ / (MOUNT root) ]
[🐳|🔨]
[🐳|🔨]   sent 1.68G bytes  received 721.86K bytes  197.30M bytes/sec
[🐳|🔨]   total size is 1.68G  speedup is 1.00
[🐳|🌿] Copying files to [ /boot (MOUNT /boot) ]
[🐳|🔨]   Internal hashtable error: illegal key supplied!
[🐳|🔨]   rsync error: errors with program diagnostics (code 13) at hashtable.c(88) [generator=3.2.7]
[🐳|🔨]   rsync error: received SIGUSR1 (code 19) at main.c(1613) [Receiver=3.2.7]
[🐳|💥] Error 13 occurred in main shell [ at /armbian/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
        run_host_command_logged() --> lib/functions/logging/runners.sh:193
   create_image_from_sdcard_rootfs() --> lib/functions/image/rootfs-to-image.sh:80
                do_with_logging() --> lib/functions/logging/section-logging.sh:81
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:104
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[🐳|💥] Cleaning up [ please wait for cleanups to finish ]

### How to reproduce?

Build on MacOS using compile.sh which in turn uses docker.

Tested with 1f69720656212154a09fa9a842e323265597a55d; v24.11 works just fine.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rpardini, 2025-02-18:
> this is through armbian's docker image

Oh, then it warrants some investigation -- sorry for the early closure

the-snork, 2025-02-18:
but anyway, let me recheck with an updated rsync in my host

rpardini, 2025-02-18:
There's indeed too much caching happening in the Docker build - https://github.com/armbian/docker-armbian-build/pull/5 - @igorpecovnik could you either merge or give me permissions there? I've more changes later.

igorpecovnik, 2025-02-18:
added permissions.

rpardini, 2025-02-19:
Heh. Un-cached builds are failing due to arm64-on-amd64 qemu throwing segmentation faults. My guess? GH doesnt want people doing crossbuilds -- I had planned to rewrite those jobs to split over arm64 and amd64 GH runners and then join the resulting Docker images in a separate step "later", but I guess it will have to be "sooner". I'll try to make time ASAP

rpardini, 2025-02-20:
I've had to split the Docker build matrix across arch'es (and join images later per-release), but finally we have updated Docker builds for both arches and all releases:
- https://github.com/armbian/docker-armbian-build/actions/runs/13435003482

@the-snork can you pull the latest image (`docker pull ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest`) and retry? We have `rsync` v `3.2.7-0ubuntu0.22.04.4` now.

rpardini, 2025-02-20:
ps: `./compile.sh` auto-pulls the image every day so manual pullng is only necessary if already tried today.

rpardini, 2025-02-20:
> My guess? GH doesnt want people doing crossbuilds

No, [known problem](https://github.com/docker/buildx/issues/1169), I solved by using native arm64 runners.

the-snork, 2025-02-20:
@rpardini - works for me; thanks.

the-snork, 2025-02-25:
forgot close it.... sorry.

closed 2025-02-25 by the-snork