armbian/build #7862: [Task]: v24.11.1 was never tagged

awinner1, 2025-02-21:
### Task description

version v24.11.1 was never tagged in the repo, please add it
- [ ] 
- [ ] 
- [ ] 
- [ ]

EvilOlaf, 2025-02-22:
I suggest to build from `main` instead. 25.02 is approaching fast and for the other my bet is that some external sources are broken already in the mean time.

igorpecovnik, 2025-02-22:
Tag:
https://github.com/armbian/build/releases/tag/v24.11.1
Branch:
https://github.com/armbian/build/tree/v24.11

But maintainability remains a problem. We freeze sources, but not packages versions - they are coming directly from Debian  / Ubuntu repo. If they drop some package, build source needs adjustment. This is very rare event, but possible.

closed 2025-04-05 by EvilOlaf