armbian/build #7888: [Feature Request]: Allow installation of mesa-vpu extension after deployment

EvilOlaf, 2025-02-28:
### Which feature would you like to have?

From time to time I recommend people toi use an image with pre-installed _mesa-vpu_ extension. Though, also more or less reasonable, they don't want to start over just to have this. For now all I can tell is 'this is the source code what the extension does, try to follow its steps' which isn't ideal.

So maybe it is possible to modify it in a way to allow running on a running Armbian image directly on hw.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

defencedog, 2025-02-28:
I second the idea ...

igorpecovnik, 2025-02-28:
I had this in mind too, but there is a mess of tasks in my mind ;) 

This fits into armbian-config, under Desktop department. I think 1st step is transferring the issue there and second, implementing it. 

> I second the idea ...

I am happy to help you adding this, even that will costs more of my time then just adding it. It is actually not that difficult. Its "just" porting from extension to https://docs.armbian.com/Contribute/Armbian-config/

EvilOlaf, 2025-06-22:
I think @amazingfate knows best

amazingfate, 2025-06-23:
> Sounds like a good possible use case of APA. Who knows about what this extension does in detail? Please ping me so we can implement this in APA together.

mesa-vpu deals with multi kernel famlies and distros:
For legacy family, it will add two ubuntu ppa: panfork-mesa for gpu panfork driver and rockchip-multimedia for vpu drivers
For vendor family, it will:
- enable a devicetree overlay panthor-gpu to enable mainline panthor gpu driver
- add ppa kisak for mainline mesa 24.1 if mesa version in ubuntu repo is lower than 24.1
- install mesa from bookworm-backports for bookworm
- add ppa rockchip-muiltimedia for vpu drivers

If ppa rockchip-multimedia is added, the extension will pin the packages from this ppa.