armbian/build #7890: [Bug]: Trixie Minimal / IOT image on rpi5 misses resize2fs

conversy, 2025-02-28:
### What happened?

Hi,

[I hope this is the correct project to create such issue]

After booting and setting up Trixie Minimal / IOT image on rpi5 I get this:
`Warning: a reboot is needed to finish resizing the filesystem`

After several reboot, the message is still here, and since I cannot install a few packages, I guess the resize is not done.

[This](https://paste.next.armbian.com/otomilowuz) log suggests that resize2fs is missing. Installing e2fsprogs and rebooting makes it work.

Best regards.

### How to reproduce?

Download and install Trixie Minimal / IOT image on rpi5

Build date: Build Date: Feb 24, 2025
SHA: fcc29f9788d4efedae61515932b9492f9a30ef1fe63f44203c56f3f655a8adf3

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 13 Trixie

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.next.armbian.com/otomilowuz

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-02-28:
Thank you for your report. It seems some dependency fall apart as this package used to be here. Fixing it ASAP. Thanks!

EvilOlaf, 2025-02-28:
Add `e2fspros` here? https://github.com/armbian/build/blob/main/config/cli/common/main/packages

igorpecovnik, 2025-02-28:
yes

closed 2025-03-01 by igorpecovnik