armbian/build #7891: [Bug]: Temporary failure resolving 'archive.ubuntu.org' when building a docker image

JohnTheCoolingFan, 2025-02-28:
### What happened?

Trying to build an image to test changes I've made, The docker image failed to build.

Additionally, I did `docker pull ghcr.io/armbian/docker-armbian-build:armbian-ubuntu-jammy-latest` and tried again, which still had the "temporary failure resolving ..." errors, but proceeded to build the image without doing that. Branch is based on `main` commit 0b88561ec332114404ff8075ab6bc2419ca66a47.

Resolving the failed hosts on the host system with `resolvectl`, `nslookup` and `dig` succeeds. Also launching a docker container with an interactive shell and resolving there works, as well as executing the apt-get commands from Dockerfile in that container.

### How to reproduce?

`./compile.sh BOARD=... BRANCH=...`

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.next.armbian.com/winisuhaxe

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rpardini, 2025-02-28:
Hey @JohnTheCoolingFan -- I don't remember handling any resolv configuration either in the docker image build, or in Armbian docker usage at all. I mean, not remembering might be just a sign of my decaying brain -- but this sounds like there's either a Docker problem or a Jammy problem here?

A few hundred CI jobs using Docker also haven't failed...

leggewie, 2025-03-16:
why am I reading of archive.ubuntu.ORG in the title?  is that what you are actually trying to access?  then that is the problem.  ubuntu is .COM

@JohnTheCoolingFan, please reconfirm

JohnTheCoolingFan, 2025-03-16:
@leggewie sorry, I got confused. The failing domain is .com, which is what it says in the log. I'll update the issue title.

JohnTheCoolingFan, 2025-03-20:
The error isn't happening anymore, so closing

closed 2025-03-20 by JohnTheCoolingFan