armbian/build #7896: [Bug]: NetworkManager-wait-online.service not enabled when NM used - means services that require working network start prematurely and fail

daniel-ayers, 2025-02-28:
### What happened?

I have observed this issue running Armbian v25.2.1 for Tinker Board running Armbian Linux 6.12.16-current-rockchip (Ubuntu noble server image).

This image uses NetworkManager for network config.  When NM is used the NetworkManager-wait-online.service must be enabled so that network-online.target is not reached until NetworkManager has brought up all the interfaces.  Service units that require the network to be up before starting rely on network-online.target working correctly otherwise they will likely fail on boot (which is the problem I observed).

See [the systemd documentation](https://systemd.io/NETWORK_ONLINE/), in particular the first two FAQs.

I encountered this problem when trying to set up a dhcp server, using the isc-dhcpd-server package.  I correctly configured the dhcp server, enabled its unit and restarted it.  It worked. However when I rebooted the Pi the dhcp server failed to start claiming "there was no subnet definition for end0". I figured out that this was because at the time isc-dhcp-server was starting the network had not yet come up, so dhcpd could not obtain the IP address & netmask for the interface to match with the subnet definition in the config file.

Reading the systemd documentation (linked above) I ran the suggested command to see if the wait services were enabled:

```
$ systemctl is-enabled NetworkManager-wait-online.service systemd-networkd-wait-online.service
disabled
disabled
```

This reveals that NetworkManager-wait-online.service is disabled, when it should be enabled when NM is used to configure the network.

I enabled NetworkManager-wait-online.service and then rebooted. The dhcp server started correctly.

### How to reproduce?

In a running Armbian system execute the following command (from the systemd documentation) and verify that the correct wait unit (depending on how network configuration is done in the image) is enabled:

```
$ systemctl is-enabled NetworkManager-wait-online.service systemd-networkd-wait-online.service
disabled
enabled

```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rpardini, 2025-03-01:
Hey @daniel-ayers, having fought the -wait-online stuff for years, while I see your point clearly, enabling it by default will cause hangs during boot for people without an ethernet connection. Specially on the first boot, we don't wanna hang on boot. Also there's many folks on wifi only.

I do think there's a good opportunity to (ask to?) enable it during the first-run "wizard". wdyt?

igorpecovnik, 2025-03-01:
Probably somewhere after this?
https://github.com/armbian/build/blob/main/packages/bsp/common/usr/lib/armbian/armbian-firstlogin#L793

conditions:
- connected wired network
- network manager stack

igorpecovnik, 2025-03-07:
Should be fixed for the future.

closed 2025-03-07 by igorpecovnik

torte71, 2025-03-12:
As @rpardini predicted, [that change](https://github.com/armbian/build/commit/7f32a5853513745ee6aa55b6d3856ac67ec036e2#diff-51f3b6ab36d40032222710dd98b635bba2b9885aee5aea801638f9495fcb02b7R148) means, that boards with only WiFi hang after entering the first root password. One has to wait about 90secs for the *-wait-online.service to time out, before being able to continue passwort entry.

Maybe moving the "systemctl start" stuff before the first password query and printing a warning, that network-init may take 1-2 minutes, is an option? So the user is at least prepared for the delay.

ig3, 2025-03-16:
The template for NetworkManager.service has had NetworkManager-wait-online as an Also in its Install section since March 2018:

```
[Install]
WantedBy=multi-user.target
Also=NetworkManager-dispatcher.service

# We want to enable NetworkManager-wait-online.service whenever this service
# is enabled. NetworkManager-wait-online.service has
# WantedBy=network-online.target, so enabling it only has an effect if
# network-online.target itself is enabled or pulled in by some other unit.
Also=NetworkManager-wait-online.service
```

Given this, if NetworkManager is enabled then NetworkManager-wait-online should be enabled. If an image has NetworkManager enabled but NetworkManager-wait-online not enabled, something has gone wrong with the setup of the image. While enabling NetworkManager-wait-online in armbian-firstlogin will fix the error, it would be better to fix the image build so it doesn't have this misconfiguration.