armbian/build #7899: [Bug]: u-boot cache version is wrong for fx-blox rk1

mahdichi, 2025-03-02:
### What happened?

I want to build Armbian V25.02 for FxBlox RK1 board but it seems the u-boot artifact is old 
here's the build log

artifact [ uboot :: uboot() ]
[🌱] Producing new & caching [ GIT_INFO_UBOOT ]
[🌱] Fetching SHA1 of 'branch' 'refs/heads/next-dev-v2024.10' [ https://github.com/radxa/u-boot.git ]
[🌱] SHA1 of branch refs/heads/next-dev-v2024.10 [ 'baac584011842f7f141d7828e311b3e098f3472e' ]
[🌱] Found cached git version [ 6f94da945f65dcddc740b514e709a620944e4389 ]
[🌱] Getting ORAS manifest [ ORAS manifest from ghcr.io/armbian/os/uboot-fxblox-rk1-vendor:2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a ]
[🌱] Artifact is available in remote cache [ ghcr.io/armbian/os/uboot-fxblox-rk1-vendor:2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a - 'uboot-fxblox-rk1-vendor - 2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a - version "2017.09" git revision "6f94da945f65dcddc740b514e709a620944e4389" patches hash "fe5f309dcfe0b0e9" Extension hooks hash "01ba4719c80b6fe911b091a7c05124b64eeece964e09c058ef8f9805daca546b" uboot functions hash "451362ceb4598378ce7ddf421de083face7a2f1eb7c717294aebaafa7cae6fa1" variables hash "db8dec9e3029a73ac95338eabd11730e2910e815279b2781c99312b21298601b" framework bash hash "b7035b95c8004ab0" - type: deb' ]
[🌱] Obtaining artifact from remote cache [ ghcr.io/armbian/os/uboot-fxblox-rk1-vendor:2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a into linux-u-boot-fxblox-rk1-vendor_2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a_arm64.deb ]
✓ Pulled      linux-u-boot-fxblox-rk1-vendor_2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a_arm64.deb                                                                              1.05/1.05 MB 100.00%  491ms
  └─ sha256:d6f0b828ee19ac0b81d21015ac33dba762358bff7c988148878d2604904f19ce
✓ Pulled      application/vnd.oci.image.manifest.v1+json                                                                                                                        1.18/1.18 kB 100.00%     0s
  └─ sha256:0fec7e8fe4a5bbbd1c40fd1680d34ab379dd2835a2c6d872f59b2dc1fc7751d0
[🔨]   Pulled [registry] ghcr.io/armbian/os/uboot-fxblox-rk1-vendor:2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a
[🔨]   Digest: sha256:0fec7e8fe4a5bbbd1c40fd1680d34ab379dd2835a2c6d872f59b2dc1fc7751d0
[💖] artifact [ obtained from remote cache: uboot-fxblox-rk1-vendor 2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a ]
[🌱] Reversioning package [ re-version 'uboot-fxblox-rk1-vendor(deb)::2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a' to '25.2.1' ]
[🔨]   removed '/mnt/sdb1/armbian/build/output/packages-hashed/global/linux-u-boot-fxblox-rk1-vendor_2017.09-S6f94-Pfe5f-Hd0b5-Vdb8d-Bb703-R448a_arm64.deb'

### How to reproduce?

./compile.sh build BOARD=fxblox-rk1 BRANCH=vendor BUILD_DESKTOP=yes CLEAN_LEVEL=oldcache,make-uboot DESKTOP_APPGROUPS_SELECTED='3dsupport browsers chat desktop_tools editors email internet multimedia office remote_desktop' DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base ENABLE_EXTENSIONS=mesa-vpu EXPERT=yes KERNEL_BTF=no KERNEL_CONFIGURE=no RELEASE=noble SHARE_LOG=yes

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-03-02:
Tried  `ARTIFACT_IGNORE_CACHE=yes` to force rebuild?

mahdichi, 2025-03-02:
> Tried `ARTIFACT_IGNORE_CACHE=yes` to force rebuild?

I will try this. Thanks.

rpardini, 2025-03-18:
I can't see how "the artifact is old?" is a problem. Prove me wrong.

leggewie, 2025-04-27:
@mahdichi Is there anything left to be done here?

closed 2025-04-28 by mahdichi