armbian/build #7907: [Bug]: Armbian changes user's global git config; fails if it can't

kjkent, 2025-03-04:
### What happened?

Running `./compile` fails if the user's global git config is a symlink to a read-only filesystem, as it usually is in NixOS. I imagine this holds true for other situations in which a user's git config is read-only to that user.

Log:
```Shell
λ ./compile.sh
error: could not lock config file /home/kjkent/.config/git/config: Read-only file system
[💥] Error 255 occurred in main shell [ at /home/kjkent/dev/armbian/lib/functions/logging/runners.sh:211
    run_host_command_logged_raw() --> lib/functions/logging/runners.sh:211
                    regular_git() --> lib/functions/general/git.sh:44
      git_ensure_safe_directory() --> lib/functions/general/git.sh:60
                           main() --> ./compile.sh:47
 ]
```
It seems unnecessary that a _Dockerised_ build requires access to my kernel modules (see https://github.com/armbian/build/issues/7052) and also needs to change my git config without any prompt beforehand? I've never encountered this before and it feels like sketchy behaviour, even if it's for benign reasons. Here's the relevant code:

```Shell
## lib/functions/general/git.sh:60
# workaround new limitations imposed by CVE-2022-24765 fix in git, otherwise  "fatal: unsafe repository"
function git_ensure_safe_directory() {
  if [[ -n "$(command -v git)" ]]; then
    local git_dir="$1"
    if [[ -e "$1/.git" ]]; then
      display_alert "git: Marking all directories as safe, which should include" "$git_dir" "debug"
      git config --global --get safe.directory "$1" > /dev/null || regular_git config --global --add safe.directory "$1"
    fi
  else
    display_alert "git not installed" "a true wonder how you got this far without git - it will be installed for you" "warn"
  fi
}
```
There is that call to `alert`, but, as seen in the output above, there's no console output other than the error.

I'm not knowledgeable about the build process enough to answer "why on earth is this function needed"; however, silently altering global git config in order to bypass a security feature that's there _specifically to plug a CVE_ does pose the question of its appropriateness.

### How to reproduce?

1. Have global user git config (eg ~/.gitconfig) be a symlink to a file on a read-only filesystem.
2. Run `./compile` in repo directory.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-03-04:
> It feels like sketchy behaviour

Indeed. This is known dirty workaround, but have to scratch my head to understand why we had to use this. IIRC it was due to some changes introduced by Git client. We tried to address this several times, but then if fail out of urgent things.

kjkent, 2025-03-05:
Makes sense. "repos not being owned by the current user" perhaps might be more of an issue on Windows, as when I searched this the other day, that appeared to be a common thread. It's not an issue I've encountered on Linux (as far as I remember!)

rpardini, 2025-03-18:
see https://github.com/armbian/build/commit/50bfbf8f0c9eaac7eeadf36fbde587e25a03e83b#commitcomment-153935903

closed 2026-10-03 by igorpecovnik