armbian/build #7912: [Bug]: uboot-patch fails to generate patch file

ig3, 2025-03-05:
### What happened?

I am attempting to create a patch for u-boot with command `$ ./compile.sh nanopi-r2s-plus uboot-patch DEBUG=yes`, with 'userpatches/config-nanopi-r2s-plus.conf':

```
BOARD=nanopi-r2s-plus
BRANCH=current
BUILD_DESKTOP=no
BUILD_MINIMAL=yes
KERNEL_BTF=no
KERNEL_CONFIGURE=no
RELEASE=bookworm
```

The compile script runs until it prompts me to make my changes and press ENTER.

In another window, I added a new file in configs with ownership and permissions consistent with existing files (i.e. owned by root.root and rw-rw-r--).

I pressed ENTER in the window running compile.sh.

The script produced output that appears to indicate an empty patch file:

```
Press ENTER to show a preview of your patch, or type 'stop' to stop patching...
[🌱] OK, here's how your diff looks like [ showing patch diff ]
[🐛] DIR_BATCAT is not set, using default [ batcat ]
[🐛] Running batcat [ batcat version 0.25.0 ]
[🐛] Running batcat bat 0.25.0 (25f4f96) [ batcat ]
[🐛] Calling batcat [ COLUMNS: 60 | --file-name /home/ian/ig3/armbian-build/output/patch/u-boot-rockchip64-current.patch - ]
───────┬────────────────────────────────────────────────────
       │ File: /home/ian/ig3/armbian-build/output/patch/
       │ u-boot-rockchip64-current.patch   <EMPTY>
───────┴────────────────────────────────────────────────────
[🐛] wait_for_disk_sync: after running batcat [ sync start, timeout_seconds:30 ]
[🐛] wait_for_disk_sync: after running batcat [ sync done worked: 1 sync_timeout: 0 ]
[🚸] Are you happy with this patch? [ Type 'yes' to accept, 'stop' to stop patching, or anything else to keep patching ]
Are you happy with the diff above? Type 'y' or 'yes' to accept, 'stop' to stop patching, anything else to keep patching: stop
```

So I stopped (i.e. I entered 'stop'). Then I checked to confirm nothing had been produced in the output/patches directory.

What I expected is that a patch file would be produced for the creation of the config file I had created.

In previous attempts I created other files. I have not attempted to change any existing files because for the change I am trying to implement I don't need to change existing files.

### How to reproduce?

Check out armbian/build branch main, commit 533fc7dca598d8019c8231c4583e80341cf6ceb5 (Feb 20 2025 - so reasonably recent)

Run the 'uboot-patch' command via compile.sh and add a file in the relevant u-boot config directory.

[log-uboot-patch-bc158496-89bb-4c22-9c3e-5bb8ceed0dac.log](https://github.com/user-attachments/files/19082507/log-uboot-patch-bc158496-89bb-4c22-9c3e-5bb8ceed0dac.log)

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-03-05:
https://forum.armbian.com/topic/50256-patch-u-boot/#comment-213866

> Alright, there seems to be a cosmetic bug. New files aren't shown for whatever reason in the patchfile  preview. However if you press y to finish up patching the patchfile in the output folder looks just fine.

ig3, 2025-03-05:
I hadn't tried accepting the patch. Having done so, I confirm - a patch file is produced with the content I was expecting.

Thanks for the quick response.

rpardini, 2025-03-18:
I've a feeling this is a bug but I can't really reproduce it. Any detailed instructions you can give me?

EvilOlaf, 2025-03-19:
Did not re-test but basically I just used `kernel-patch` with the parameters above, created two files with nonsense contents (check patchfile here: https://forum.armbian.com/topic/50256-patch-u-boot/#comment-213866). Preview was empty, patch was fine.

EvilOlaf, 2025-03-19:
I get your point. Though I assume there are other (valid?) reasons to have a new file created from a patch.

rpardini, 2025-03-19:
https://chatgpt.com/share/67dad256-d7f4-800c-90fc-863962628c99

ig3, 2025-03-28:
With patch #7979 merged, from my perspective, this issue is resolved.

@rpardini You mentioned that you want to discourage people making a null-patch. I don't know what you mean by this. From your perspective, does this change promote the production of null patches when files are added?

rpardini, 2025-03-28:
"adding files using a patch" == null patch

the patch file will read

```patch
--- /dev/null
+++ b/some/new/file
```

We've mechanisms (`dt/` folder in kernels, `defconfigs/` folder in some u-boots) to avoid null patches; but not all u-boots use them. See https://github.com/armbian/build/issues/7772

rpardini, 2025-03-28:
But yeah since we have #7772 go ahead and close this.

ig3, 2025-03-28:
Thanks for the explanation of 'null-patch'. The uboot-patch command could be modified to do the right thing with new files, or better documented so that it is clear how and when to use it. My conclusion is that I was abusing it and therefore encountered some inconsistent behaviour. #7979 makes the behaviour more consistent but does not prevent or discourage the inappropriate use: it is an improvement but not ideal. None the less, I will close this issue, leaving further improvement of the command and documentation to another day.

closed 2025-03-28 by ig3