armbian/build #7940: [Bug]: debootstrap permission denied

merll002, 2025-03-10:
### What happened?

After compiling the kernel, I got this error: `debootstrap: 1866: cannot create /home/leo/build/.tmp/rootfs-099f19d9-fb54-46ff-9802-99076bd9950e/test-dev-null: Permission denied`

The .tmp folder is owned by root.

I am compiling in a Podman container.

### How to reproduce?

Self explanatory

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.next.armbian.com/ovohewegoz

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-03-11:
Not sure if podman is supported. Have you tried regular docker or on bare metal?

merll002, 2025-03-11:
Ok, I will try that. Is there any way I can avoid rebuilding the kernel as that finished successfully?

EvilOlaf, 2025-03-11:
If it has build successfully and the artifacts was cached correctly it should pull it from local cache.

merll002, 2025-03-12:
Well this issue hasn't been fixed, rather, a workaround to use docker has been proposed. It's up to you guys if you want to close this!

EvilOlaf, 2025-03-13:
Well if there was never official support for podman, then there never was an issue.
https://docs.armbian.com/Developer-Guide_Build-Preparation/#requirements

merll002, 2025-03-13:
I'll close then!

closed 2025-03-13 by merll002