armbian/build #7942: [Task]: Add sector size support to partitioning for UFS

HeyMeco, 2025-03-12:
### Task description

While working on Radxa Nio 12L support which uses 4096 byte native storage via UFS I noticed just setting offset isn’t enough as the build system always defaults to 512 byte's with the included sfdisk from the docker image.
- [ ] Find a solution to have --sector-size enabled for sfdisk in our build process which is a thing in newer versions. (See https://man7.org/linux/man-pages/man8/sfdisk.8.html)
- [ ] Update partitioning.sh to make use of sector sizes in `sfdisk` and `losetup`
- [ ] Add documentation for sector size env variable

HeyMeco, 2025-03-12:
Support for the --sector-size command is implemented after util-linux 2.42.

amazingfate, 2025-03-12:
2.41-rc1 has included it, packaged in debian experimental: https://packages.debian.org/source/experimental/util-linux

HeyMeco, 2025-03-12:
We could try and see if manually installing 2.41 to the docker image introduces any bugs in our build process?

HeyMeco, 2025-04-08:
Need to test it out but it's likely this PR for Qualcomm https://github.com/armbian/build/pull/8053 could fix this for Mediatek too

- [ ] Test and confirm

dogblade-png, 2025-04-20:
I think I might a solution to your issue. Check it out here [click](https://forreststonesolutions.com/robots/)

closed 2025-09-16 by HeyMeco