armbian/build #7988: [Bug]: There is no support for Rockchip NPU

easy-and-simple, 2025-03-22:
### What happened?

As Armbian OS aswell Armbian build framework doesn't support Rockchip NPU for Rock 3A
On Armbian OS for Radxa Rock 3A board Rockchip NPU is not supported. The same for Armbian build framework.
In Armbian OS i check it by different commands. In Armbian build framework I cant find option to enable NPU in kernel options

### How to reproduce?

Believe on my word.
Also search kernel options in framework or use dmesg command in OS

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

easy-and-simple, 2025-03-23:
However, the problem lies within the framework itself, not with any external drivers or third-party components. The issue is a flaw in the framework's code, which prevents the NPU driver from being included in the build image for the RK3568 CPU or the Radxa Rock 3A board.
If the NPU driver had been included in the framework configuration for RK35886 CPU boards like the Radxa Rock 3A board, it would have been a driver or third-party issue if there were any NPU-related problems caused by bugs in the driver code.

igorpecovnik, 2025-03-23:
> which prevents the NPU driver from being included

We appreciate your interest in our project. At this time, we can only include features that have been fully developed. While additional features are possible with financial support, the current level of donations covers less than 0.5% of our work. What you receive now is already a freely provided contribution.

If the current project does not meet your needs, you are welcome to explore other options. Additionally, since this is open-source software, you have the freedom to fork the project and implement any features you need.

Thank you for your understanding and support!

closed 2025-03-23 by igorpecovnik