armbian/build #7995: [Feature Request]: Are you planning to support the armel architecture?

zooyer, 2025-03-24:
### Which feature would you like to have?

Some older devices are armel and want to install Armbian.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

leggewie, 2025-03-24:
I guess it all comes down to volunteers stepping up to build and test.  Are you interested to do that,@zooyer?

zooyer, 2025-03-25:
> I guess it all comes down to volunteers stepping up to build and test. Are you interested to do that,[@zooyer](https://github.com/zooyer)?

I am very willing to participate in the test。

igorpecovnik, 2025-03-25:
> willing to participate in the test

Thank you for your willingness! However, from our perspective, the situation is far more complex.  

Such idea / project would place additional pressure on multiple people / teams, including:  
- Hardware developers  
- Build framework developers  
- Test automation engineers  
- Coordinators managing communication with testers  (additional support beyond you is necessary)  
- Support staff  

This is a considerable undertaking, and unfortunately, we lack dedicated funding sources beyond private contributions. Industry is not interested for old hardware, meaning reliance must be placed on third-party organizations. [Donations](https://github.com/sponsors/armbian) from end users traditionally cover only 0.5% of the total project costs, making it an unsustainable funding model for supporting (endless) feature requests.

tl;dr;  

Armbian's infrastructure can provide some assistance, but not at scale. To move forward, we strongly encourage you to build momentum within the community forums and rally a critical mass of contributors. This will require significant effort and time, with no guaranteed outcome - however, this is the approach I would take.  

While we can support your initiative through promotion via our social channels, we do not have the capacity for direct involvement. Even maintaining critical Armbian infrastructure is a challenge with our current resources. One of critical parts is testing infrastructure - Linux development (especially embedded) in 2025 does not rely on users to tests, but robots.

zooyer, 2025-03-25:
> Thank you for your willingness! However, from our perspective, the situation is far more complex.
>
> Such idea / project would place additional pressure on multiple people / teams, including:
>
> * [...]

Thank you for your response! I understand the challenges involved. I'll start by exploring the community and seeing how I can gather support.

Santiagoelao, 2025-04-20:
I think I might a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)

tabrisnet, 2025-09-07:
https://www.debian.org/releases/trixie/ suggests that `armel` is a dead letter.
do you mean `armhf` or `armv7`?