armbian/build #7996: [Bug]: Building cloud image fails at Grub

igorpecovnik, 2025-03-24:
### What happened?

Cloud images making seems to be broken.

### How to reproduce?

```./compile.sh build BETA=no BOARD=uefi-x86 BRANCH=current BUILD_DESKTOP=no BUILD_MINIMAL=yes ENABLE_EXTENSIONS=cloud-init IMAGE_VERSION=25.2.3 RELEASE=bookworm SHARE_LOG=yes ```

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/poviqusino

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rpardini, 2025-03-24:
Broken implies it ever worked -- I don't think the version that got merged (eg, not mine) ever worked with grub. It's meant for SBCs only.

igorpecovnik, 2025-03-24:
Aha, I was trying to improve cloud images, x86 / arm64, with grub, without armbian-firmware, perhaps with lean kernel ... cloud-init support would be handy to have by default, so I tried the existing.

leggewie, 2025-03-25:
certainly, it would be great to have an agreed-to standard for testing builds on virtual compute

igorpecovnik, 2025-03-26:
Actually this here is not a problem, so closing.

> certainly, it would be great to have an agreed-to standard for testing builds on virtual compute

I started here:
https://github.com/armbian/build/pull/8008
but it is too early for review. Just to see the direction:
- cloud image has dedicated kernel with minimal hw support, only what is needed for VPS
- boot time is around 1s
- dedicated configs for docker, zram, ...

closed 2025-03-26 by igorpecovnik