armbian/build #8023: [Bug]: Rock Pi S compiles `rk3308-s0-ext-antenna` dto even tho it is not compatible with this board

Jorropo, 2025-03-28:
### What happened?

With my RockPi-S I go to `armbian-config` overlay selection and I enable `s0-ext-antenna` however this shows up while booting:
```
Applying kernel provided DT overlay rk3308-s0-ext-antenna.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
226 bytes read in 6 ms (36.1 KiB/s)
```
Doing git archeology this happens because the dts for `s0-ext-antenna` target symbols like `board_antenna` that are defined in `board-rocks0-0001-deviceTree.patch` but not in `rk3308-rock-pi-s.dts` (file in upstream linux).
However looking at the hardware the wifi setup for the rock-pi-s0 and rock-pi look different:

On the Rock-Pi-S0, we can clearly spot an RF Switch IC (tiny black SMD component in the following screenshot), with impedance matched traces going to the Ceramic antenna, the ufl connector and the SOM:
[image]

However on the Rock-Pi-S there is no such switch, or built-in ceramic antenna to begin with, we can spot the Realtek WiFi+BT IC, a π matching network and impedance matched traces straight to the antenna connector:
[image]
Critically I do not see an RF Switch IC or a builtin ceramic antenna.

So if my understand is correct this file should not be compiled to begin with given there is no built-in antenna to begin with on the Rock Pi S.

### How to reproduce?

.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Jorropo, 2025-03-28:
Hum, this is a bit bad because trying to apply this incompatible dto make all other following dto fail with `FTD_ERR_BADMAGIC` because it seems uboot continues in a corrupted state after trying to apply `rk3308-s0-ext-antenna`.

It were an unfun time to debug why my spidev weren't showing up and the answer ended being this.

dogblade-png, 2025-04-20:
I may I found a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)