armbian/build #8031: [Bug]: RockPi-E not working ethernet and fail boot 24.11.1/25.2.1

Benik3, 2025-03-30:
### What happened?

On Rock-Pi E v1.2 booting 24.11.1 or 25.2.1 minimal/IOT Noble image fail with "illegal free".
Booting 25.5.0-trunk.316_Rockpi-e_plucky_current_6.12.21_minimal works with not working ethernet.

Upgrading 24.8.3 to 6.12 kernel ends also with not working ethernet, like when booting 25.5 trunk image.
This issue is probably after 24.11.1 version, because I have one board upgraded from 24.8.3 to 24.11.1 and it works OK.

Here is successful boot log of 24.8.3: https://paste.armbian.com/ixuvimofis.sql
here is boot log of 24.8.3 upgraded to 6.12: https://paste.armbian.com/ahezanixol.sql
Here is failing boot of 25.2.1:  https://paste.armbian.com/onuhumomob.yaml 
Here is boot log of 25.5_trunk.316 with verbosity=7:  https://paste.armbian.com/bimosoyupi.yaml

### How to reproduce?

boot fail: Write 25.2.1 image to sd card and boot from it.
eth fail: Write 25.5 trunk image on sd card and boot from it, or upgrade 24.8.3 image

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-03-30:
This issue tracker does only accept issues with the build framework, _building_ Armbian to say.
For issues _running_ please refer to forums.

closed 2025-03-30 by EvilOlaf

igorpecovnik, 2025-03-30:
Rather ask / proceed here: https://forum.armbian.com/forum/247-radxa-rock-pi-e/

Benik3, 2025-03-30:
OK, I understand. It's already on the forum, but there was no response, so I didn't know if anyone noticed it:
https://forum.armbian.com/topic/50489-rockpi-e-doesnt-work-with-kernel-612/

igorpecovnik, 2025-03-30:
> but there was no response

Two weeks is nothing. Use workaround: https://docs.armbian.com/User-Guide_Armbian-Config/System/#install-alternative-kernels

You can try here: https://bugzilla.kernel.org/ Timeframe From reporting, resolving and fix integration is few months in best case. You can also hire someone to fix this bug for you and share the fix.