armbian/build #8052: [Bug]: libraspberrypi-bin E: Unable to correct problems, you have held broken packages.

sashasimkin, 2025-04-07:
### What happened?

When trying to build rpi4b image I receive the following error:
```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libraspberrypi-bin : Depends: libraspberrypi0 (= 0~20230913+gitcc1ca18-0ubuntu2) but it is not going to be installed
                      Depends: libc6 (>= 2.38) but 2.36-9+deb12u10 is to be installed
E: Unable to correct problems, you have held broken packages.
```
I've checked the bookworm apt contents and libc6 is fixed there at 2.36:
```
Package: libc6
Version: 2.36-9+deb12u10
...
APT-Sources: http://deb.debian.org/debian bookworm/main arm64 Packages
```

### How to reproduce?

Run ./compile.sh for rpi4b, kernel: current

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-04-07:
full logs missing

igorpecovnik, 2025-04-07:
Debian building looks like broken. Let me fix this when possible. Building Ubuntu image will work.

sashasimkin, 2025-04-07:
Hey @igorpecovnik, thank you!

Right now I have a dirty workaround to fix older version of these libs in `config/sources/families/bcm2711.conf`.

```
function pre_install_distribution_specific__add_rpi_packages() {
        # If we're skipping the Armbian repo, and on Debian, we'll install Debian version of Wifi/BT-related firmware.
        if [[ "${SKIP_ARMBIAN_REPO}" == "yes" ]]; then
                display_alert "SKIP_ARMBIAN_REPO is set to" "${SKIP_ARMBIAN_REPO} - no Armbian repo packages will be installed" "info"
                if [[ "${DISTRIBUTION}" == "Debian" ]]; then
                        display_alert "Installing Debian-sourced Wifi/BT firmware packages" "Debian::bcm2711::${BOARD}" "info"
                        do_with_retries 3 chroot_sdcard_apt_get_update
                        chroot_sdcard_apt_get_install firmware-brcm80211 firmware-misc-nonfree bluez-firmware bluetooth
                fi
                if [[ "${DISTRIBUTION}" == "Ubuntu" ]]; then
                        display_alert "Installing Ubuntu-sourced Wifi/BT firmware packages plus raspi-config and raspi-eeprom" "Ubuntu::bcm2711::${BOARD}" "info"
                        do_with_retries 3 chroot_sdcard_apt_get_update
                        chroot_sdcard_apt_get_install rpi-eeprom pi-bluetooth busybox raspi-config bluetooth
                fi
        else
                # we store Rpi firmware packages in our repository
                # https://github.com/armbian/os/wiki/Import-3rd-party-packages
                display_alert "Enable Armbian repository to fetch Rpi packages" "" "info"
                mv "${SDCARD}"/etc/apt/sources.list.d/armbian.sources.disabled "${SDCARD}"/etc/apt/sources.list.d/armbian.sources
                do_with_retries 3 chroot_sdcard_apt_get_update
                chroot_sdcard_apt_get_install libraspberrypi0=0~20220324+gitc4fd1b8-0ubuntu1~22.04.1 libraspberrypi-bin=0~20220324+gitc4fd1b8-0ubuntu1~22.04.1
                chroot_sdcard apt-mark hold libraspberrypi-bin
                chroot_sdcard_apt_get_install rpi-eeprom linux-firmware-raspi pi-bluetooth busybox raspi-config
                ## disable armbian repository
                mv "${SDCARD}"/etc/apt/sources.list.d/armbian.sources "${SDCARD}"/etc/apt/sources.list.d/armbian.sources.disabled
        fi
}
```

Unfortunately, I couldn't manage to do that with an extension, because for some reason `"${SDCARD}"/etc/apt/sources.list.d/armbian.sources.disabled` seems to not exist in my extension.

i.e. with this extension:
```
function pre_install_distribution_specific__01-add_rpi_packages() {
    # !!!
    # https://github.com/armbian/build/issues/8052
    # latest version is 0~20230913+gitcc1ca18-0ubuntu2 which depends on libc >= 2.38
    # Bookworm main has version 2.36-9+deb12u10
    # !!!
	# If we're skipping the Armbian repo, and on Debian, we'll install Debian version of Wifi/BT-related firmware.
	if [[ "${SKIP_ARMBIAN_REPO}" == "yes" ]]; then
		display_alert "SKIP_ARMBIAN_REPO is set to" "${SKIP_ARMBIAN_REPO} - no Armbian repo packages will be installed" "info"
		if [[ "${DISTRIBUTION}" == "Debian" ]]; then
			display_alert "Installing Debian-sourced Wifi/BT firmware packages" "Debian::bcm2711::${BOARD}" "info"
			do_with_retries 3 chroot_sdcard_apt_get_update
			chroot_sdcard_apt_get_install firmware-brcm80211 firmware-misc-nonfree bluez-firmware bluetooth
		fi
		if [[ "${DISTRIBUTION}" == "Ubuntu" ]]; then
			display_alert "Installing Ubuntu-sourced Wifi/BT firmware packages plus raspi-config and raspi-eeprom" "Ubuntu::bcm2711::${BOARD}" "info"
			do_with_retries 3 chroot_sdcard_apt_get_update
			chroot_sdcard_apt_get_install rpi-eeprom pi-bluetooth busybox raspi-config bluetooth
		fi
	else
		# we store Rpi firmware packages in our repository
		# https://github.com/armbian/os/wiki/Import-3rd-party-packages
		display_alert "Enable Armbian repository to fetch Rpi packages" "" "info"
		mv "${SDCARD}"/etc/apt/sources.list.d/armbian.list.disabled "${SDCARD}"/etc/apt/sources.list.d/armbian.list
		do_with_retries 3 chroot_sdcard_apt_get_update
		chroot_sdcard_apt_get_install rpi-eeprom linux-firmware-raspi pi-bluetooth libraspberrypi-bin=0~20230913+gitcc1ca18-0ubuntu2 busybox raspi-config
		chroot_sdcard apt-mark hold libraspberrypi-bin

		## disable armbian repository
		mv "${SDCARD}"/etc/apt/sources.list.d/armbian.list "${SDCARD}"/etc/apt/sources.list.d/armbian.list.disabled
	fi
}
```

I receive the following output:
```
[🌱] Enable Armbian repository to fetch Rpi packages
mv: cannot stat '/home/ubuntu/build/.tmp/rootfs-69d4e7c1-fa46-4b6f-975e-2b55a515cb23/etc/apt/sources.list.d/armbian.list.disabled': No such file or directory
[💥] Error 1 occurred in main shell [ at /home/ubuntu/build/userpatches/extensions/my-extensions.sh:30
   pre_install_distribution_specific__01-add_rpi_packages() --> userpatches/extensions/my-extensions.sh:30
   pre_install_distribution_specific() --> <extension_magic>:263
          call_extension_method() --> lib/functions/general/extensions.sh:57
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:22
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
```

sashasimkin, 2025-04-10:
Hi @igorpecovnik , I saw that the build here https://github.com/armbian/os/pull/309/files succeeded and it was merged.

Unfortunately it didn't help - the error is the same:

```
[🐛] Extra envs for apt: [ 'DEBIAN_FRONTEND=noninteractive' ]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
busybox is already the newest version (1:1.35.0-4+b3).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libraspberrypi-bin : Depends: libraspberrypi0 (= 0~20230913+gitcc1ca18-0ubuntu2) but it is not going to be installed
                      Depends: libc6 (>= 2.38) but 2.36-9+deb12u10 is to be installed
E: Unable to correct problems, you have held broken packages.
[🐛] Size of apt/deb cache after 'apt-get --no-install-recommends install rpi-eeprom linux-firmware-raspi pi-bluetooth libraspberrypi-bin busybox raspi-config' [ 478951417 bytes ]
[🐛] Local apt cache size unchanged after 'apt-get --no-install-recommends install rpi-eeprom linux-firmware-raspi pi-bluetooth libraspberrypi-bin busybox raspi-config' [ at 478951417 bytes ]
[🐛] Unmounting apt deb cache dir [ /home/ubuntu/build/.tmp/rootfs-05aeb0fb-1362-4797-8ef3-5645828d3425/var/cache/apt ]
[🐛] Unmounting apt list cache dir [ /home/ubuntu/build/.tmp/rootfs-05aeb0fb-1362-4797-8ef3-5645828d3425/var/lib/apt/lists ]
[💥] Error 1 occurred in main shell [ at /home/ubuntu/build/lib/functions/logging/runners.sh:15
   chroot_sdcard_apt_get_install() --> lib/functions/logging/runners.sh:15
   pre_install_distribution_specific__add_rpi_packages() --> config/sources/families/bcm2711.conf:238
   pre_install_distribution_specific() --> <extension_magic>:263
          call_extension_method() --> lib/functions/general/extensions.sh:57
         build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:22
   full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36
          do_with_default_build() --> lib/functions/main/default-build.sh:42
         cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25
        armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136
                 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176
                           main() --> compile.sh:50
 ]
[💥] Cleaning up [ please wait for cleanups to finish ]
```

It seems that RPI folks have pushed incompatible version for bookworm into their repositories :(

igorpecovnik, 2025-04-10:
Yeah, i know ... this https://github.com/armbian/os/actions/runs/14386405108 hasn't finished, then it has to sync to servers. I hope it will be ok now.

igorpecovnik, 2025-04-11:
This is fixed, one more to go.
https://github.com/armbian/build/issues/8069

closed 2025-04-11 by igorpecovnik

sashasimkin, 2025-04-11:
aah, TIL. thank you so much! I just checked and the build for armbian is back working 🎉

sashasimkin, 2025-04-14:
Indeed. I seem to have had some packages cached.

Fresh install & build on 24.04 of armbian bookwork failed with the following :(

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 pi-bluetooth : Depends: bluez-firmware but it is not installable
                Depends: raspberrypi-sys-mods but it is not installable
E: Unable to correct problems, you have held broken packages.
[🐛] Size of apt/deb cache after 'apt-get --no-install-recommends install rpi-eeprom linux-firmware-raspi pi-bluetooth busybox raspi-config' [ 388829071 bytes ]
[🐛] Local apt cache size unchanged after 'apt-get --no-install-recommends install rpi-eeprom linux-firmware-raspi pi-bluetooth busybox raspi-config' [ at 388829071 bytes ]
```

igorpecovnik, 2025-04-14:
Yes. I am on it but its quite a mess. Possible workaround is SKIP_ARMBIAN_REPO=yes

sashasimkin, 2025-04-14:
It seems to fix the issue, thank you!

igorpecovnik, 2025-04-14:
> It seems to fix the issue, thank you!

Proper fix coming with a PR.