armbian/build #8056: [Bug]: K3: SK-AM68 cannot boot with Armbian U-Boot

HeyMeco, 2025-04-08:
### What happened?

My red pcb SK-AM68 (should be the same to the blue ones) doesn't boot Armbian images.
Replacing the u-boot with the one provided from TI on their Yocto images `U-Boot 2024.04-ti-ga970f6e51043` makes my SK-AM68 boot Armbian while the 2023 & 2025 K3 U-Boot's we currently use result in:

```
U-Boot 2025.01-armbian-2025.01-S6d41-P0000-H67e9-V9004-Bb703-R448a (Mar 13 2025 - 04:18:10 +0000)
SoC:   J721S2 SR1.0 HS-FS
Model: Texas Instruments J721S2 EVM
Board: AM68-SK-SOM rev E2
DRAM:  16 GiB
i2c_write: error waiting for data ACK (status=0x116)
pca953x gpio@22: Error reading output register
Core:  104 devices, 32 uclasses, devicetree: separate
Flash: 0 Bytes
MMC:   mmc@4f80000: 0, mmc@4fb0000: 1
Loading Environment from nowhere... OK
In:    serial@2880000
Out:   serial@2880000
Err:   serial@2880000
Net:   eth0: ethernet@46000000port@1
Hit any key to stop autoboot:  0 
i2c_write: error waiting for data ACK (status=0x116)
pca953x gpio@22: Error reading output register
Card did not respond to voltage select! : -110
```

This is how it looks on the vendor copied u-boot:

```
U-Boot 2024.04-ti-ga970f6e51043 (Nov 13 2024 - 14:26:23 +0000)

SoC:   J721S2 SR1.0 HS-FS
Model: Texas Instruments AM68 SK
Board: AM68-SK-SOM rev E2
DRAM:  2 GiB (effective 16 GiB)
Core:  104 devices, 32 uclasses, devicetree: separate
Flash: 0 Bytes
MMC:   mmc@4fb0000: 1
Loading Environment from nowhere... OK
In:    serial@2880000
Out:   serial@2880000
Err:   serial@2880000
am65_cpsw_nuss ethernet@46000000: K3 CPSW: nuss_ver: 0x6BA02102 cpsw_ver: 0x6BA82102 ale_ver: 0x00293904 Ports:1
Net:   eth0: ethernet@46000000port@1
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1 is current device
SD/MMC found on device 1
574 bytes read in 23 ms (23.4 KiB/s)
Loaded env from uEnv.txt
Importing environment from mmc1 ...
21029376 bytes read in 236 ms (85 MiB/s)
103376 bytes read in 26 ms (3.8 MiB/s)
Working FDT set to 88000000
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
Working FDT set to 88000000
   Loading Device Tree to 000000008fee3000, end 000000008fffffff ... OK
Working FDT set to 8fee3000

Starting kernel ...
```

@Grippy98 @glneo 

### How to reproduce?

Try to boot Armbian 24.11.1 or self built images from build system.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Grippy98, 2025-04-08:
@HeyMeco Since I can't reproduce that bug on my end for boot, can you try builds with k3.conf changed to use- 

1. Newer uboot - declare -g BOOTBRANCH="tag:v2025.04"

2. Using TI U-Boot - 
BOOTSOURCE=https://github.com/TexasInstruments-Sandbox/ti-u-boot 
BOOTBRANCH="tag:v2025.01"

Any difference?

HeyMeco, 2025-04-08:
@Grippy98 

1. Newer U-Boot has the same issue
2. ti-u-Boot 2025.01 also with the issue
3. Using ti-u-boot with branch 2024.01 it boots.

soborakso, 2025-04-20:
I believe I might a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)