armbian/build #8061: [Feature Request]: PPPoe Module - BPI F3

andersonwallacerj, 2025-04-09:
### Which feature would you like to have?

I'm trying to recompile the kernel to enable it to create my personal network but I can't.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

EvilOlaf, 2025-04-09:
> but I can't.

Cannot reproduce due to lack of information.
Moved since it has nothing to do with configng

EvilOlaf, 2025-04-09:
Seems like Docker and plain Bookworm build host fail. Plain Noble seems good. gcc too old in former?

andersonwallacerj, 2025-04-09:
> Seems like Docker and plain Bookworm build host fail. Bare metal Noble

Yeah, I have both situations.
No docker to install and pppoe module isn't enable into the kernel.

I tried to recompilethe kernel, but none of the ones I tried I was able to
compile without errors.... 😢

andersonwallacerj, 2025-04-09:
> Official supported build host is Ubuntu Noble, which is used by most
> people and where our automation quickly find possible troubles and we fix
> them ASAP. It might works elsewhere too, but we don't pay attention nor
> support that. We simply can't due to limited resources.

I'm having trouble to write different images with SD card or writing
directly to the emm, I use an "old" armbian 23 or 24 and update it until
the noble one.

I'll try again tomorrow. 😉

igorpecovnik, 2025-04-09:
> I'm having trouble to write different images with SD card

I hope you do not use Etcher as its broken. Not just for Armbian, but many others.

Good luck.

EvilOlaf, 2025-04-11:
@pyavitz can you reproduce that building f3 works with plain Noble only? Docker/Jammy/ seems broken.

pyavitz, 2025-04-11:
> [@pyavitz](https://github.com/pyavitz) can you reproduce that building f3 works with plain Noble only? Docker/Jammy/ seems broken.

I personally use Noble (aarch64) as my HOST. x86_64 should also work fine. Jammy is a no go with this.

closed 2025-04-19 by igorpecovnik

karbun01, 2025-04-20:
I think I might a solution to your issue. Check it out here [click](https://forreststonesolutions.com/robots/)