armbian/build #8066: [Feature Request]: Enable eBPF + CO-RE for Rockchip/rk3588 builds

gclawes, 2025-04-10:
### Which feature would you like to have?

Support eBPF + CO-RE via for rockchip/rk3588 builds.  `DEBUG_INFO_BTF=y` is set in upstream [debian](https://salsa.debian.org/kernel-team/linux/-/blob/debian/latest/debian/config/arm64/config?ref_type=heads#L2574)/[ubuntu](https://launchpad.net/ubuntu/+source/linux-raspi/6.8.0-1009.10}, it's needed for pretty much any modern eBPF tool for Compile Once-Run Everywhere.  It would be nice to have it automatically in kernel updates, rather than having to recompile the kernel every update to maintain a feature that's usually set in upstream distros.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2025-04-11:
Already done, https://github.com/armbian/build/pull/7629

gclawes, 2025-04-14:
Is #7629 only for current/edge kernels, or vendor as well?  The image/board I'm testing with is Turing RK1 (`Armbian_community_25.5.0-trunk.370_Turing-rk1_bookworm_vendor_6.1.99_minimal.img`)

I see mention of outdated vendor kernels in that PR: https://github.com/armbian/build/pull/7629#issuecomment-2594659169

rpardini, 2025-04-14:
There was some back and forth, but it should be enabled thanks to some heroics by AF.

Please run `cat /boot/config-6.1.99-vendor-rk35xx | grep BTF` or similar.

gclawes, 2025-04-14:
Hmm, latest image I downloaded from `armbian/community` is [25.5.0-trunk.370](https://github.com/armbian/community/releases/tag/25.5.0-trunk.370), and I don't see it there:
```
root@nas:/tmp# losetup -l
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE                                                                             DIO LOG-SEC
/dev/loop0         0      0         0  0 /tmp/Armbian_community_25.5.0-trunk.370_Turing-rk1_bookworm_vendor_6.1.99_minimal.img   0     512
root@nas:/tmp# mount | grep loop0
/dev/loop0p1 on /mnt/test type ext4 (rw,relatime)
root@nas:/tmp# grep BTF /mnt/test/boot/config-6.1.99-vendor-rk35xx
CONFIG_VIDEO_SONY_BTF_MPX=m
```

Latest `armbian/build` release is [v25.5.0-trunk.385](https://github.com/armbian/build/releases/tag/v25.5.0-trunk.385), should I expect to see it there if I build that myself (or wait for it to land in `armbian/community` releases)?

gclawes, 2025-04-14:
Yeah, confirmed I see it when building the latest image:
```
gclawes@nas[±|tag:v25.5.0-trunk.385 {1} ✓]:~/repos/github.com/armbian/build $ losetup -l
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE                                                                                                                                 DIO LOG-SEC
/dev/loop0         0      0         0  0 /home/gclawes/repos/github.com/armbian/build/output/images/Armbian-unofficial_25.05.0-trunk_Turing-rk1_bookworm_vendor_6.1.99_minimal.img   0     512
gclawes@nas[±|tag:v25.5.0-trunk.385 {1} ✓]:~/repos/github.com/armbian/build $ mount | grep /dev/loop0
/dev/loop0p1 on /mnt/test type ext4 (rw,relatime)
gclawes@nas[±|tag:v25.5.0-trunk.385 {1} ✓]:~/repos/github.com/armbian/build $ grep BTF /mnt/test/boot/config-6.1.99-vendor-rk35xx
CONFIG_VIDEO_SONY_BTF_MPX=m
CONFIG_DEBUG_INFO_BTF=y
CONFIG_PAHOLE_HAS_SPLIT_BTF=y
CONFIG_DEBUG_INFO_BTF_MODULES=y
# CONFIG_MODULE_ALLOW_BTF_MISMATCH is not set
```

rpardini, 2025-04-20:
Nice, lets close this then?

juggerindentity, 2025-04-20:
I believe I might a solution to your issue. Check it out here [click](https://forreststonesolutions.com/robots/)

gclawes, 2025-04-20:
Yep, thanks for the help!

closed 2025-04-20 by gclawes