armbian/build #8069: [Bug]: RPI on Debian fails to assemble due to packaging issue

igorpecovnik, 2025-04-11:
### What happened?

[🐳|🔨]   Unpacking armbian-bsp-cli-rpi4b-current (25.05.0-trunk) ...
[🐳|🔨]   dpkg: error processing archive /root/armbian-bsp-cli-rpi4b-current_25.05.0-trunk_arm64__1-PCe1b9-Vd907-Hb756-B1e5d-Rf36a.deb (--unpack):
[🐳|🔨]    trying to overwrite '/etc/kernel/postinst.d/z50-raspi-firmware', which is also in package raspi-firmware 1.20220830+ds-1

### How to reproduce?

Build Debian variant.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/ivohuturog

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leggewie, 2025-04-13:
Have you tried "Replaces"?  Might be a hack that masks the real issue.  But "Replaces" is there for exactly this kind of situation.

To dig deeper: Are both deb packages necessary?  Do they both need to provide the same file?  I am trying to undestand what the root of the problem is.

armbianworker, 2025-04-13:
>  I am trying to undestand what the root of the problem is.

That makes two of us :) We have same bsp package, but different upstream firmware packages. Ubuntu and Debian have different methods for Rpi and we are in the middle ... currently don't know what is the best path.

igorpecovnik, 2025-04-14:
Works now, closing.

closed 2025-04-14 by igorpecovnik

guilherme-custela, 2025-04-15:
I still get the error 💥] Error 1 occurred in main shell [ at /home/inspire/build/lib/functions/logging/runners.sh:15

 chroot_sdcard_apt_get_install() --> lib/functions/logging/runners.sh:15

 pre_install_distribution_specific__add_rpi_packages() --> config/sources/families/bcm2711.conf:250

 pre_install_distribution_specific() --> <extension_magic>:256

 call_extension_method() --> lib/functions/general/extensions.sh:57

 build_rootfs_and_image() --> lib/functions/main/rootfs-image.sh:22

 full_build_packages_rootfs_and_image() --> lib/functions/main/default-build.sh:36

 do_with_default_build() --> lib/functions/main/default-build.sh:42

 cli_standard_build_run() --> lib/functions/cli/cli-build.sh:25

 armbian_cli_run_command() --> lib/functions/cli/utils-cli.sh:136

 cli_entrypoint() --> lib/functions/cli/entrypoint.sh:176

 main() --> compile.sh:50

 ]

[💥] Cleaning up [please wait for cleanups to finish]

EvilOlaf, 2025-04-15:
Also make sure your cloned build framework is up to date (`git pull`).

leggewie, 2025-08-11:
> Have you tried "Replaces"? Might be a hack that masks the real issue. But "Replaces" is there for exactly this kind of situation.
> 
> To dig deeper: Are both deb packages necessary? Do they both need to provide the same file? I am trying to undestand what the root of the problem is.
> 
> > Would it be enough to simply rename this file and check inside if Ubuntu distribution is present on run-time (in postinst script) as we only need to execute it there.
> 
> Don't do this. No need for this hackery.

@igorpecovnik I am currently reviewing #8240 but to do that, I need to understand what is going on.  You never fully explained the situation.  Why are we not using Replaces?  What is the real situation here?  You pushed a "solution" but it created an issue for @rpardini.  A lot of people are here saying they don't understand what is going on.  Kindly explain.