armbian/build #8075: [Bug]: linux-image-current-meson64 lacks 8189fs kernel module

alive4ever, 2025-04-12:
### What happened?

With the update to kernel version `6.12.22-current-meson64`, the `8189fs` module is no longer included.

My repurposed set top box which is based on `gxl-p212` has Realtek 8188FTV sdio card which works with 8189fs but not with 8189es.

[I have built out of tree kernel module](https://github.com/alive4ever/rtl8189fs-armbian-current-meson64.git) and load it successfully on the current stable kernel.

Additional info:

```
apt show linux-image-current-meson64
Package: linux-image-current-meson64
Version: 25.2.3
Priority: optional
Section: kernel
Source: linux-6.12.22
Maintainer: Armbian Linux <info@armbian.com>
Installed-Size: 279 MB
Provides: linux-image, linux-image-armbian, armbian-current
Armbian-Kernel-Version: 6.12.22
Armbian-Kernel-Version-Family: 6.12.22-current-meson64
Armbian-Original-Hash: 6.12.22-S5576-Dc54f-P43c4-C83f1H71d6-HK01ba-Ve377-B9bbb-R448a
Download-Size: 49.9 MB
APT-Manual-Installed: yes
APT-Sources: http://apt.armbian.com bookworm/main arm64 Packages
Description: Armbian Linux current kernel image 6.12.22-current-meson64
 This package contains the Linux kernel, modules and corresponding other files.
 version "6.12.22" git revision "55767d6e74ef155f43fff978ad4e2f69f3f8fad7" codename "Baby Opossum Posse" drivers hash "c54fcdb1_3bddffba" patches hash "43c4a574c6e080b2" .config hash "83f14d54dcc723c2" .config hook hash "71d6d9e120b6949b" variables hash "e3771f6e9d9ac20167c59982d4ccc58a8e68b59ea73d40feb1c1c8172d251477" framework bash hash "9bbbd4b74c282bc5"

N: There are 6 additional records. Please use the '-a' switch to see them.
```

### How to reproduce?

From the linux-image-current-meson64 package, find file with name `8189fs.ko`.

```sh
apt download linux-image-current-meson64
dpkg-deb --contents linux-image-current-meson64_25.2.3_arm64__6.12.22-S5576-Dc54f-P43c4-C83f1H71d6-HK01ba-Ve377-B9bbb-R448a.deb | grep 8189fs
```

The file is not there.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnTheCoolingFan, 2025-04-12:
Looks like this is also the case for allwinner: https://forum.armbian.com/topic/51055-lost-wifi-after-upgrade-on-latest-bookwormnoble-images/

alive4ever, 2025-04-15:
> Looks like this is also the case for allwinner

Thanks for the quick fix.

The `8189fs` kernel module is now present within linux-image-current-meson64 in the beta.armbian.com repository, but the stable repository (apt.armbian.com) is still having this issue.

igorpecovnik, 2025-04-15:
> but the stable repository (apt.armbian.com) is still having this issue.

Updating this is a manual process that also involves a costly manual testing procedure and pressure to support if we make update from trunk. Its a big change, not just this driver. For now, updating this will have to wait until May 2024.

Since the bug has been resolved at build time—which is what this tracker is about—I'll go ahead and close it.

closed 2025-04-15 by igorpecovnik