armbian/build #8083: [Bug]: `kernel-dtb` does not output when `unset BOOT_FDT_FILE` is present

EvilOlaf, 2025-04-13:
### What happened?

I noticed this when preparing https://github.com/armbian/build/pull/8082.
When `unset BOOT_FDT_FILE` was present there was no output of the preprocessed dts files in `output/` folder.
Once commented it worked as expected.

Not sure if this expected or an actual bug.

### How to reproduce?

- set  `unset BOOT_FDT_FILE` in board config file.
- use `kernel-dtb` for the same board and - if applicable - branch
- check `output`

Haven't tested with other boards/branches though.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.next.armbian.com/reyawolagi

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rpardini, 2025-04-13:
There's no way kernel-dtb can know what to output if that is unset...

EvilOlaf, 2025-04-14:
Alright, kind of expected this. As mentioned wasn't sure if it would output something regardless.
Though maybe this should be a fatal error then rather than a warning.

`--> (35) WARNING: Board 'orangepi5-max' branch 'edge' [ No BOOT_FDT_FILE set for board, skipping further processing ]`

saucekantrol, 2025-04-20:
I believe I might a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)