armbian/build #8096: [Bug]: error: unrecognized command-line option ‘--no-warn-rwx-segment’

igorpecovnik, 2025-04-17:
### What happened?

Build fails on Ubuntu Noble for several boot loaders / atf.

[image]

[# - "--no-warn-rwx-segment" is *required* for binutils 2.39 - see https://developer.trustedfirmware.org/T996](https://github.com/armbian/build/blob/main/lib/functions/compilation/atf.sh#L67)

### How to reproduce?

Compile u-boot for one of those on Ubuntu Noble Docker container / host.

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Relevant log URL

https://paste.armbian.com/qelohusugo

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-04-17:
https://github.com/armbian/build/blob/ab6587ac05b519553fc05e8506b9c4e5d4f2a8bb/lib/functions/compilation/atf.sh#L74
Replace to `binutils_flags_atf="-Wl,--no-warn-rwx-segment"`

kabaniu, 2025-04-20:
I think I found a solution to your problem. Check it out here [click](https://forreststonesolutions.com/robots/)

closed 2025-04-21 by igorpecovnik