armbian/build #8120: [Feature Request]: Draft proposal: “Mainline” kernel / U-Boot build target for Armbian

timsurber, 2025-04-25:
### Which feature would you like to have?

I’d like to propose introducing an additional build flavour—tentatively called “mainline”—that tracks the vanilla Linux kernel and U-Boot, built strictly without any Armbian-specific patches.

**Why this makes sense**
- Easier upgrades: Staying patch-free  eliminates the merge conflicts that routinely appear when rebasing our downstream trees onto new kernel versions, so we can move to newer releases faster and with less effort.
- Upstream-verified code only: Every line would have passed review on the LKML or U-Boot mailing list, giving us the best possible security and stability baseline.
- First-class dev platform: Maintainers and contributors who are working on getting features upstream—or who want to reproduce issues against vanilla kernels—would have an official, ready-made image.

**Trade-offs**

The obvious downside is coverage: many boards currently enabled by Armbian rely on out-of-tree patches, so the mainline target will initially support a smaller subset of devices.

**Next steps**

If the idea resonates, we could:

  -  Define a minimal board list that already runs on upstream only.
   - Add a BUILD_FLAVOUR=mainline option to the build system that skips the patch stages.
   -  Publish nightly images and gather feedback from developers before advertising the flavour to a wider audience.

I’d be happy to refine the details and help with the implementation if the community agrees this is worth pursuing.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

lanefu, 2025-04-25:
I've often thought it would be nice to have a vanilla mainline kernel like you're describing.     There's different levels of purity to consider such as keeping the convenience of device tree overlays etc.   I think it's quite achievable overall 

I think generically solving for uboot (for well supported boards) would probably still be fairly involved and would definitely treat as a separate project

timsurber, 2025-04-25:
Thank you for your answer. Yes, limiting this proposal to just the kernel would be fine for me.

rpardini, 2025-04-26:
> There's different levels of purity to consider

"Pure mainline" gold standard, or not supported.

Indeed, but _many_ boards could get a bootable image with zero patching, and just by knowing the u-boot `defconfig` name and (original) Armbian `LINUXFAMILY` (read: kernel name). I think most rockchip would just work and possibly also some sunxi?

We could experiment with a `BOARD` like `generic-mainline-rk64` that takes a `DEFCONFIG=` param and uses hooks to twist enough knobs to avoid all patching & create a new `LINUXFAMILY=rk64-mainline`. We could even have `BRANCH=current` be the latest stable, and `edge` against `linux-next`.

rpardini, 2025-04-26:
Of course I forgot one needs to know the SoC (`BOOT_SOC`) so we know which blobs to use, but we've code that parses that out of the defconfig-name string. Dunno if that applies to sunxi. For `meson64` with it's FIP blobs it would be a bit harder.

misuzu, 2025-04-27:
Upstream u-boot with default settings supports generic [/boot/extlinux/extlinux.conf](https://docs.u-boot.org/en/latest/develop/distro.html) (and [UEFI](https://wiki.nixos.org/wiki/NixOS_on_ARM/UEFI)) out of the box so it's possible to make a single generic os image. 

The main issue is getting the u-boot itself to be loaded by the board. Each board has its own build of u-boot and it could be handled separately from the os image, like bios upgrade on x86_64.

The easiest way is to flash board specific u-boot to board spi. If spi is unavailable then it has to be burned to disk. There're different ways to do this depending on the board so each board needs a guide on how to flash u-boot (e.g. like [this](https://wiki.nixos.org/wiki/NixOS_on_ARM/Allwinner/GPT_Installation) or [this](https://wiki.nixos.org/wiki/NixOS_on_ARM/Radxa_ROCK_4)).

timsurber, 2025-04-28:
Thank you for your answers so far :)

After reading your comments I thought about creating a new `KERNEL_TARGET` `mainline` and add it to the config for each fully mainline supported board.
This will execute a hook which will modify `KERNELPATCHDIR` to unset it or set it to an empty directory.

The rest of the build with regards to the bootloader and blobs will execute as defined by the board config.

I think this should integrate nicely into the existing build framework.

What are your thoughts on this?

timsurber, 2025-05-05:
simple implementation in #8149

closed 2025-05-05 by timsurber