armbian/build #8153: DKMS: '.gnu.linkonce.this_module section size must match the kernel's built struct module size at run time' error

epsilonrt, 2025-05-05:
I'm having a problem installing a driver. Once compiled correctly, the system gives me an error. Note that I performed the same procedure with RaspiOS on an arm64 board with the same kernel version, and it worked. I also tried with Debian 12 on a PC, and it worked. The problem therefore lies with Armbian. Also I have been using tty0tty on Armbian for a very long time and this problem just appeared.

The driver I want to install is https://github.com/lcgamboa/tty0tty.

After two days of searching, I decided to ask for help. I'm having trouble loading the module on a 6.12.20 kernel running aarch64 (Armbian on a NanoPi Neo Core2 board). As we can see the kernel version of the driver is the same as that of the kernel.

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ modinfo tty0tty
filename:       /lib/modules/6.12.20-current-sunxi64/updates/dkms/tty0tty.ko
license:        GPL
description:    tty0tty null modem driver
author:         Luis Claudio Gamboa Lopes <lcgamboa@yahoo.com>
depends:
name:           tty0tty
vermagic:       6.12.20-current-sunxi64 SMP mod_unload aarch64
epsilonrt@nanopineocore2:~/src/tty0tty/module$ uname -a
Linux nanopineocore2 6.12.20-current-sunxi64 #1 SMP Sat Mar 22 19:54:28 UTC 2025 aarch64 GNU/Linux
```

The scripted installation seems to be working correctly (there is no MOK generated by the dkms package under Armbian...):

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ sudo ./dkms-install.sh
About to run dkms install steps...
Deprecated feature: REMAKE_INITRD (/usr/src/tty0tty-1.4/dkms.conf)
Creating symlink /var/lib/dkms/tty0tty/1.4/source -> /usr/src/tty0tty-1.4
Sign command: /lib/modules/6.12.20-current-sunxi64/build/scripts/sign-file
Binary /lib/modules/6.12.20-current-sunxi64/build/scripts/sign-file not found, modules won't be signed
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVERSION=6.12.20-current-sunxi64 KERNELDIR=/lib/modules/6.12.20-current-sunxi64/build......
Cleaning build area...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

tty0tty.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.20-current-sunxi64/updates/dkms/
depmod...
Finished running dkms install steps.
echo "tty0tty" > /etc/modules-load.d/tty0tty.conf
cp 99-tty0tty.rules  /etc/udev/rules.d/
udevadm control --reload-rules && udevadm trigger
chown root:root /etc/udev/rules.d/99-tty0tty.rules
```

However the module does not load and `dmesg` tells me:

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ dmesg -T | tail
[...]
[dim. mai  4 17:37:11 2025] module tty0tty: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
```

It seems that the size of the descriptor structure doesn't match the kernel version.
A loading attempt with modprobe indeed indicates a problem, even when trying to force it:

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ sudo modprobe tty0tty
modprobe: ERROR: could not insert 'tty0tty': Exec format error
epsilonrt@nanopineocore2:~/src/tty0tty/module$ sudo modprobe tty0tty --force-vermagic
modprobe: ERROR: could not insert 'tty0tty': Exec format error
epsilonrt@nanopineocore2:~/src/tty0tty/module$ sudo modprobe tty0tty -f
modprobe: ERROR: could not insert 'tty0tty': Exec format error
```

However, the vermagic of my module corresponds to the version of my kernel...

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ modinfo tty0tty
filename:       /lib/modules/6.12.20-current-sunxi64/updates/dkms/tty0tty.ko
license:        GPL
description:    tty0tty null modem driver
author:         Luis Claudio Gamboa Lopes <lcgamboa@yahoo.com>
depends:
name:           tty0tty
vermagic:       6.12.20-current-sunxi64 SMP mod_unload aarch64
epsilonrt@nanopineocore2:~/src/tty0tty/module$ uname -a
Linux nanopineocore2 6.12.20-current-sunxi64 #1 SMP Sat Mar 22 19:54:28 UTC 2025 aarch64 GNU/Linux
```

I did some research, reinstalled the linux-headers package, and did the same thing on a Raspberry Pi 4B with the same kernel version and architecture:

```
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ sudo ./dkms-install.sh
About to run dkms install steps...
Deprecated feature: REMAKE_INITRD (/usr/src/tty0tty-1.4/dkms.conf)
Creating symlink /var/lib/dkms/tty0tty/1.4/source -> /usr/src/tty0tty-1.4
Sign command: /lib/modules/6.12.20+rpt-rpi-v8/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVERSION=6.12.20+rpt-rpi-v8 KERNELDIR=/lib/modules/6.12.20+rpt-rpi-v8/build....
Signing module /var/lib/dkms/tty0tty/1.4/build/tty0tty.ko
Cleaning build area...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

tty0tty.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.20+rpt-rpi-v8/updates/dkms/
depmod....
Finished running dkms install steps.
echo "tty0tty" > /etc/modules-load.d/tty0tty.conf
cp 99-tty0tty.rules  /etc/udev/rules.d/
udevadm control --reload-rules && udevadm trigger
chown root:root /etc/udev/rules.d/99-tty0tty.rules
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ sudo modprobe tty0tty
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ sudo dmesg | tail
[...]
[30201.664164] tty0tty: loading out-of-tree module taints kernel.
[30201.668963] tty0tty null modem driver v1.4
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ ls /dev/tnt*
/dev/tnt0  /dev/tnt1  /dev/tnt2  /dev/tnt3  /dev/tnt4  /dev/tnt5  /dev/tnt6  /dev/tnt7
```

It works!!

```
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ uname -a
Linux rpi-bookworm-arm64 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
epsilonrt@rpi-bookworm-arm64:~/src/tty0tty/module $ modinfo tty0tty
filename:       /lib/modules/6.12.20+rpt-rpi-v8/updates/dkms/tty0tty.ko.xz
license:        GPL
description:    tty0tty null modem driver
author:         Luis Claudio Gamboa Lopes <lcgamboa@yahoo.com>
srcversion:     C65C31D2ED5FD58973628A0
depends:
name:           tty0tty
vermagic:       6.12.20+rpt-rpi-v8 SMP preempt mod_unload modversions aarch64
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        6F:11:CE:4E:78:9F:15:76:4C:EF:26:2D:03:B4:11:D8:8C:A9:BE:1B
sig_hashalgo:   sha512
signature:      63:C1:54:78:B4:01:C5:5F:A3:84:B8:9B:36:C7:19:93:A5:98:FC:37:
                B6:23:BE:33:01:04:E5:C0:E4:0A:BB:70:C9:86:94:00:5F:69:D2:DA:
                9A:4D:40:9F:DF:54:3F:FF:B6:AA:D8:04:6D:D7:E0:40:66:CC:37:31:
                40:43:25:E3:A0:75:1E:3A:2D:22:B8:F2:42:A6:75:D7:37:42:27:44:
                DE:77:4F:76:6B:24:6A:00:01:43:AF:00:02:9B:F1:CA:E4:9B:01:53:
                0A:11:CF:8A:E0:63:EC:DD:BC:5A:A3:C8:E8:4F:4C:DC:B4:C4:E4:5A:
                90:F2:10:4B:EC:91:BF:C5:F1:04:91:24:88:E5:ED:EE:65:65:98:0D:
                50:8A:4A:7A:79:E5:C5:E5:36:37:3F:C1:F9:ED:67:42:BE:DB:5C:45:
                84:34:19:06:AD:C0:65:2D:71:18:0A:C0:F7:CE:AD:57:1B:42:B1:CA:
                7B:CD:57:CF:E1:00:EC:F4:A2:61:E3:8F:11:72:C6:87:5B:4B:26:8F:
                9D:E2:10:70:C4:9A:E5:AE:04:8C:BE:9C:6E:29:41:CC:66:72:A7:12:
                9F:26:DF:30:DF:97:9D:1D:F0:E6:F8:A2:47:59:80:0A:C1:47:6D:7D:
                5C:6A:C8:3F:C1:C2:6C:9F:1D:78:D6:CD:42:A6:10:07
```

However, the driver is signed, even though it's not on Armbian. Could that be the reason?
I'm a bit lost here, any help would be appreciated, thanks.

igorpecovnik, 2025-05-05:
>  The problem therefore lies with Armbian

Most likely its kernel and kernel headers issue with particular kernel family. First try with Armbian x86 or Armbian for Raspberry Pi to confirm that. Quality of Allwinner kernel stack is not on the level of Rpi and certainly not on x86. 

This is probably not build issue per se, thus its better to bring it up in community forums. 

> I'm a bit lost here, any help would be appreciated, thanks.

No worries. Most of people are lost without looking deeper.

My 2c.

rpardini, 2025-05-06:
@epsilonrt  

1) what release (bookworm/jammy/noble/trixie/etc) are you running? the Armbian kernel is (AFAIK) built with a Jammy toolchain, so if using something very different that might fail. We haven't seen this happen since 2020, though.
2) can you get other modules working? eg `v4l2loopback-dkms` is usually a good test. if that works, but tty0tty doesn't, then it's something about that module or what it does.
3) try building your own image/kernel for the release you're running -- then try building the module on that image. Maybe that will help.

epsilonrt, 2025-05-06:
Thank you all for your help.
I'm using Bookworm with the latest image available on the Armbian website. But, I also ran the test with an image built with compile.sh on Jammy and the problem is the same.

I followed @igorpecovnik's advice and first verified that the problem also exists on a NanoPi Neo Core board (armhf architecture), and everything works on that architecture.

I'll check with Armbian on a Raspberry Pi 4B board with aarch64 architecture to see if the problem persists.

If not, I'll report the issue to the Allwinner community forum.

```
epsilonrt@nanopineo:~/src/tty0tty/module$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Armbian 25.5.0-trunk.513 bookworm
Release:        12
Codename:       bookworm
```

build and install the dkms driver:

```
epsilonrt@nanopineo:~/src/tty0tty/module$ sudo ./dkms-install.sh
[sudo] Mot de passe de epsilonrt :
About to run dkms install steps...
Deprecated feature: REMAKE_INITRD (/usr/src/tty0tty-1.4/dkms.conf)
Creating symlink /var/lib/dkms/tty0tty/1.4/source -> /usr/src/tty0tty-1.4
Sign command: /lib/modules/6.12.23-current-sunxi/build/scripts/sign-file
Binary /lib/modules/6.12.23-current-sunxi/build/scripts/sign-file not found, modules won't be signed
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVERSION=6.12.23-current-sunxi KERNELDIR=/lib/modules/6.12.23-current-sunxi/build......
Cleaning build area...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

tty0tty.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.23-current-sunxi/updates/dkms/
depmod.....
Finished running dkms install steps.
echo "tty0tty" > /etc/modules-load.d/tty0tty.conf
cp 99-tty0tty.rules  /etc/udev/rules.d/
udevadm control --reload-rules && udevadm trigger
chown root:root /etc/udev/rules.d/99-tty0tty.rules
```

load the driver and check:

```
epsilonrt@nanopineo:~/src/tty0tty/module$ sudo modprobe tty0tty
epsilonrt@nanopineo:~/src/tty0tty/module$ dmesg | tail
[...]
[ 2710.600289] tty0tty: loading out-of-tree module taints kernel.
[ 2710.605285] tty0tty null modem driver v1.4

epsilonrt@nanopineo:~/src/tty0tty/module$ ls /dev/tnt*
/dev/tnt0  /dev/tnt1  /dev/tnt2  /dev/tnt3  /dev/tnt4  /dev/tnt5  /dev/tnt6  /dev/tnt7
```

Everything works!

epsilonrt, 2025-05-06:
I just checked on a Raspberry Pi 4b with Armbian bookworn aarch64 and it works.
So the problem only exists on Allwinner sunxi64, so I'm posting the issue to the Allwinner community forum.
Thank you all for taking some of your precious time to answer me.

```
epsilonrt@r14:~/src/tty0tty/module$ sudo ./dkms-install.sh
About to run dkms install steps...
Deprecated feature: REMAKE_INITRD (/usr/src/tty0tty-1.4/dkms.conf)
Creating symlink /var/lib/dkms/tty0tty/1.4/source -> /usr/src/tty0tty-1.4
Sign command: /lib/modules/6.12.22-current-bcm2711/build/scripts/sign-file
Binary /lib/modules/6.12.22-current-bcm2711/build/scripts/sign-file not found, modules won't be signed
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

Building module:
Cleaning build area...
'make' -j4 KVERSION=6.12.22-current-bcm2711 KERNELDIR=/lib/modules/6.12.22-current-bcm2711/build....
Cleaning build area...
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/tty0tty/1.4/source/dkms.conf)

tty0tty.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.12.22-current-bcm2711/updates/dkms/
depmod....
Finished running dkms install steps.
echo "tty0tty" > /etc/modules-load.d/tty0tty.conf
cp 99-tty0tty.rules  /etc/udev/rules.d/
udevadm control --reload-rules && udevadm trigger
chown root:root /etc/udev/rules.d/99-tty0tty.rules

epsilonrt@r14:~/src/tty0tty/module$ sudo modprobe tty0tty
epsilonrt@r14:~/src/tty0tty/module$ dmesg | tail
[   15.046221] bcmgenet fd580000.ethernet end0: Link is Up - 1Gbps/Full - flow control off
[   21.678333] Bluetooth: RFCOMM TTY layer initialized
[   21.678364] Bluetooth: RFCOMM socket layer initialized
[   21.678384] Bluetooth: RFCOMM ver 1.11
[   84.092535] mempolicy: Request by 'ps' to set policy to 5 ignored
[   84.092553] mempolicy: Request by 'ps' to set policy to 3 ignored
[  261.550048] mempolicy: Request by 'ps' to set policy to 5 ignored
[  261.550065] mempolicy: Request by 'ps' to set policy to 3 ignored
[  643.744771] tty0tty: loading out-of-tree module taints kernel.
[  643.750941] tty0tty null modem driver v1.4

epsilonrt@r14:~/src/tty0tty/module$ ls /dev/tnt*
/dev/tnt0  /dev/tnt1  /dev/tnt2  /dev/tnt3  /dev/tnt4  /dev/tnt5  /dev/tnt6  /dev/tnt7
```

closed 2025-05-06 by epsilonrt

epsilonrt, 2025-05-06:
To answer @rpardini's suggestion, here's what happens with `v4l2loopback-dkms`, package installation fails, installation by compiling the sources brings the same error:

```
epsilonrt@nanopineocore2:~/src/v4l2loopback$ sudo modprobe v4l2loopback
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
```

Here are the details:

```
epsilonrt@nanopineocore2:~/src/tty0tty/module$ sudo apt install v4l2loopback-dkms
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait
Paquets suggérés :
  v4l2loopback-utils
Les NOUVEAUX paquets suivants seront installés :
  v4l2loopback-dkms
0 mis à jour, 1 nouvellement installés, 0 à enlever et 1 non mis à jour.
Il est nécessaire de prendre 37,2 ko dans les archives.
Après cette opération, 118 ko d'espace disque supplémentaires seront utilisés.
Réception de :1 http://deb.debian.org/debian bookworm/main arm64 v4l2loopback-dkms all 0.12.7-2 [37,2 kB]
37,2 ko réceptionnés en 0s (311 ko/s)
Sélection du paquet v4l2loopback-dkms précédemment désélectionné.
(Lecture de la base de données... 139085 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../v4l2loopback-dkms_0.12.7-2_all.deb ...
Dépaquetage de v4l2loopback-dkms (0.12.7-2) ...
Paramétrage de v4l2loopback-dkms (0.12.7-2) ...
Loading new v4l2loopback-0.12.7 DKMS files...
Building for 6.12.20-current-sunxi64
Building initial module for 6.12.20-current-sunxi64
Error! Bad return status for module build on kernel: 6.12.20-current-sunxi64 (aarch64)
Consult /var/lib/dkms/v4l2loopback/0.12.7/build/make.log for more information.
dpkg: erreur de traitement du paquet v4l2loopback-dkms (--configure) :
 le sous-processus paquet v4l2loopback-dkms script post-installation installé a renvoyé un état de sortie d'erreur 10
Des erreurs ont été rencontrées pendant l'exécution :
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

epsilonrt@nanopineocore2:~/src/tty0tty/module$ cat /var/lib/dkms/v4l2loopback/0.12.7/build/make.log
DKMS make.log for v4l2loopback-0.12.7 for kernel 6.12.20-current-sunxi64 (aarch64)
mar. 06 mai 2025 18:49:46 UTC
Building v4l2-loopback driver...
make -C /lib/modules/6.12.20-current-sunxi64/build M=/var/lib/dkms/v4l2loopback/0.12.7/build modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-6.12.20-current-sunxi64 »
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c: In function ‘vidioc_querycap’:
/var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.c:717:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Werror=implicit-function-declaration]
  717 |         strlcpy(cap->driver, "v4l2 loopback", sizeof(cap->driver));
      |         ^~~~~~~
      |         strncpy
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:229 : /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.o] Erreur 1
make[2]: *** [/usr/src/linux-headers-6.12.20-current-sunxi64/Makefile:1946 : /var/lib/dkms/v4l2loopback/0.12.7/build] Erreur 2
make[1]: *** [Makefile:224 : __sub-make] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-6.12.20-current-sunxi64 »
make: *** [Makefile:43 : v4l2loopback.ko] Erreur 2
```

So the package installation failed, going through the sources of the github repository:

```
epsilonrt@nanopineocore2:~/src/v4l2loopback$ make
grep: /etc/dkms/framework.conf.d/*.conf: Aucun fichier ou dossier de ce type

++++++ To sign the  module, you must set KBUILD_SIGN_KEY/KBUILD_SIGN_CERT to point to the signing key/certificate!
++++++ For your convenience, we try to read these variables as 'mok_signing_key' resp. 'mok_certificate' from /etc/dkms/framework.conf /etc/dkms/framework.conf.d/*.conf

++++++ If your certificate requires a password, pass it via the KBUILD_SIGN_PIN env-var!
grep: /etc/dkms/framework.conf.d/*.conf: Aucun fichier ou dossier de ce type
[... 14 lines cut ...]
cc   v4l2loopback-ctl.o   -o v4l2loopback-ctl
make[1] : on quitte le répertoire « /home/epsilonrt/src/v4l2loopback/utils »
epsilonrt@nanopineocore2:~/src/v4l2loopback$ sudo make install
grep: /etc/dkms/framework.conf.d/*.conf: Aucun fichier ou dossier de ce type

++++++ To sign the  module, you must set KBUILD_SIGN_KEY/KBUILD_SIGN_CERT to point to the signing key/certificate!
++++++ For your convenience, we try to read these variables as 'mok_signing_key' resp. 'mok_certificate' from /etc/dkms/framework.conf /etc/dkms/framework.conf.d/*.conf

++++++ If your certificate requires a password, pass it via the KBUILD_SIGN_PIN env-var!
grep: /etc/dkms/framework.conf.d/*.conf: Aucun fichier ou dossier de ce type
++++++ E.g. using 'export KBUILD_SIGN_PIN; read -s -p "Passphrase for signing key : " KBUILD_SIGN_PIN; sudo --preserve-env=KBUILD_SIGN_PIN make sign'

make -C /lib/modules/`uname -r`/build M=/home/epsilonrt/src/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-6.12.20-current-sunxi64 »
  INSTALL /lib/modules/6.12.20-current-sunxi64/updates/v4l2loopback.ko
  DEPMOD  /lib/modules/6.12.20-current-sunxi64
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1] : on quitte le répertoire « /usr/src/linux-headers-6.12.20-current-sunxi64 »

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)

epsilonrt@nanopineocore2:~/src/v4l2loopback$ sudo depmod
epsilonrt@nanopineocore2:~/src/v4l2loopback$ sudo modprobe v4l2loopback
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
```