armbian/build #8161: [Bug]: `BUILD_KSRC` switch gone

EvilOlaf, 2025-05-06:
### What happened?

With the update to the new framework the `BUILD_KSRC` has been removed it seems.
No clue if that was by mistake or intentionally removed due to reasons.
From time to time folks asking for a kernel source package for their board. Seemed handy.

The function to create the config used to look something like this:

```bash
#!/usr/bin/env bash
create_linux-source_package() {
	ts=$(date +%s)
	local sources_pkg_dir tmp_src_dir
	tmp_src_dir=$(mktemp -d)
	trap "ret=\$?; rm -rf \"${tmp_src_dir}\" ; exit \$ret" 0 1 2 3 15
	sources_pkg_dir=${tmp_src_dir}/${CHOSEN_KSRC}_${REVISION}_all
	mkdir -p "${sources_pkg_dir}"/usr/src/ \
		"${sources_pkg_dir}"/usr/share/doc/linux-source-${version}-${LINUXFAMILY} \
		"${sources_pkg_dir}"/DEBIAN

	cp "${SRC}/config/kernel/${LINUXCONFIG}.config" "default_${LINUXCONFIG}.config"
	xz < .config > "${sources_pkg_dir}/usr/src/${LINUXCONFIG}_${version}_${REVISION}_config.xz"

	display_alert "Compressing sources for the linux-source package"
	tar cp --directory="$kerneldir" --exclude='.git' --owner=root . |
		pv -p -b -r -s "$(du -sb "$kerneldir" --exclude=='.git' | cut -f1)" |
		pixz -4 > "${sources_pkg_dir}/usr/src/linux-source-${version}-${LINUXFAMILY}.tar.xz"
	cp COPYING "${sources_pkg_dir}/usr/share/doc/linux-source-${version}-${LINUXFAMILY}/LICENSE"

	cat <<- EOF > "${sources_pkg_dir}"/DEBIAN/control
		Package: linux-source-${version}-${BRANCH}-${LINUXFAMILY}
		Version: ${version}-${BRANCH}-${LINUXFAMILY}+${REVISION}
		Architecture: all
		Maintainer: $MAINTAINER <$MAINTAINERMAIL>
		Section: kernel
		Priority: optional
		Depends: binutils, coreutils, linux-base
		Provides: linux-source, linux-source-${version}-${LINUXFAMILY}
		Recommends: gcc, make
		Description: This package provides the source code for the Linux kernel $version
	EOF

	fakeroot dpkg-deb -b -Z${DEB_COMPRESS} -z0 "${sources_pkg_dir}" "${sources_pkg_dir}.deb"
	rsync --remove-source-files -rq "${sources_pkg_dir}.deb" "${DEB_STORAGE}/"

	te=$(date +%s)
	display_alert "Make the linux-source package" "$(($te - $ts)) sec." "info"
	rm -rf "${tmp_src_dir}"
}
```

### How to reproduce?

-

### Branch

main (main development branch)

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-05-06:
I don't recall. @rpardini

rpardini, 2025-05-06:
Building a source package makes very little sense -- the packaging is done in armbian/build for some 3 years now.
Yes, one could export a .deb containing a flat tarball (thus losing all patch information) of sources after patching, but building _that_ would result in absolutely nothing usable.
Developers in-the-know enough could use `rewrite-kernel-patches PUSH_<xxx>` to get a git tree that is patched the same way as our packages.

EvilOlaf, 2025-05-06:
Sometimes user demands don't necessarily make sense ;)

The-going, 2025-05-06:
> Building a source package makes very little sense

Oh ho ho.
Then why do all self-respecting distributions provide source packages for their binary packages?

This is not even a matter for discussion.

EvilOlaf, 2025-05-14:
Not sure if the source package should contain either the Linux version or the hash or both...couldn't manage to get it to work either way. For some reason `${version}` is empty seems empty when used but when read the variable has contents...I don't get it.