armbian/build #8171: [Feature Request]: mention Orange Pi CM4 in Orange Pi 3B profile

ssinyagin, 2025-05-09:
### Which feature would you like to have?

The latest Bookworm image for `Orange Pi 3B` has successfully booted on `Orange Pi CM4`. It will be great to have the mentioning of CM4 in the `config/boards/orangepi3b.csc`, or maybe it's worth creating a new CSC for CM4.

So far, everything looks to be working fine: WiFi and Bluetooth, USB are recognized. Also, a two-GigE baseboard by Waweshare is showing two ports, as expected.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

defencedog, 2025-05-10:
@ssinyagin  can you direct me to the link of exact debian image you used where all hardware is running?

ssinyagin, 2025-05-10:
> [@ssinyagin](https://github.com/ssinyagin) can you direct me to the link of exact debian image you used where all hardware is running?

I built it the standard way, then dd onto SD card, booted from it, then `armbian-install` copied it to the MMC:
```
./compile.sh BOARD=orangepi3b BRANCH=edge BUILD_DESKTOP=no BUILD_MINIMAL=yes KERNEL_CONFIGURE=no RELEASE=bookworm
```

myeyestore, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

eyestorefivem, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemservereyestore, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.

defencedog, 2025-05-10:
@ssinyagin Sorry for any inconvenience. Can you please upload the `.img` file as well as headers `.deb`

ssinyagin, 2025-05-10:
> [@ssinyagin](https://github.com/ssinyagin) Sorry for any inconvenience. Can you please upload the `.img` file as well as headers `.deb`

but it literally takes 10 minutes to build an image. Uploading and downloading 2GB would be slower than that

amazingfate, 2025-05-11:
>Also, a two-GigE baseboard by Waweshare is showing two ports, as expected

The mainline devicetree of opi3b should have only enabled one gmac, what's the output of command:
ls -l /sys/class/net/