armbian/build #8197: sunxi: `reboot` does not perform a reboot

djurny, 2025-05-15:
Seems that a recent rolling update has made my sunxi Orange Pi zero boards stuck at reboot. Kernel shuts down fine, no errors, prints out "Restarting system" or "Rebooting system" but it seems to just get stuck. Magic SysRq not working so kernel has left the building. Board does not actually reboot/reset. It requires a power cycle to start again.

Could be that the board just halts or powers down, but not quite sure how to determine that with for example a multimeter.

Will update the description here after I found out what rolling release version caused this to happen.

Groetjes,

djurny, 2025-05-16:
Downgraded from 25.2.3 to 24.11.1 seems to resolve the issue (includes `tio` timestamps):
```
[19:49:45] [   87.616053] systemd-shutdown[1]: Stopping MD devices.
[19:49:45] [   87.621767] systemd-shutdown[1]: All MD devices stopped.
[19:49:45] [   87.627230] systemd-shutdown[1]: Detaching DM devices.
[19:49:45] [   87.632944] systemd-shutdown[1]: All DM devices detached.
[19:49:45] [   87.638393] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[19:49:45] [   87.656493] systemd-shutdown[1]: Syncing filesystems and block devices.
[19:49:45] [   87.663469] systemd-shutdown[1]: Rebooting.
[19:49:45] [   87.720599] reboot: Restarting system

[19:49:46] U-Boot SPL 2024.01-armbian (Feb 23 2024 - 10:47:39 +0000)
[19:49:46] DRAM: 256 MiB
[19:49:46] Trying to boot from MMC1
[19:49:46] ns16550_serial serial@1c28000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

[19:49:46] U-Boot 2024.01-armbian (Feb 23 2024 - 10:47:39 +0000) Allwinner Technology
```

```
[DOWNGRADE] armbian-bsp-cli-orangepizero-current:armhf 25.2.3 -> 24.11.1
[DOWNGRADE] armbian-firmware:armhf 25.2.3 -> 24.11.1
[DOWNGRADE] linux-dtb-current-sunxi:armhf 25.2.3 -> 24.11.1
[DOWNGRADE] linux-image-current-sunxi:armhf 25.2.3 -> 24.11.1
[UPGRADE] linux-u-boot-orangepizero-current:armhf 24.2.1 -> 24.11.1
```

Note that I at first thought it was due to U-Boot, so I tried with built U-Boots and downgraded linux-u-boot-orangepizero-current packages. 

Will dive into the diffs to see if I can find the culprit.

Grt,

djurny, 2025-05-16:
In `25.2.2` `reboot` also works.

```
========================================
...
[DOWNGRADE] armbian-bsp-cli-orangepizero-current:armhf 25.08.0-trunk -> 25.2.2
[DOWNGRADE] armbian-firmware:armhf 25.08.0-trunk -> 25.2.2
[DOWNGRADE] linux-dtb-current-sunxi:armhf 25.08.0-trunk -> 25.2.2
[DOWNGRADE] linux-image-current-sunxi:armhf 25.08.0-trunk -> 25.2.2
[DOWNGRADE] linux-u-boot-orangepizero-current:armhf 25.08.0-trunk -> 25.2.2
========================================
```

djurny, 2025-05-21:
Seems like issue introduced when moving from `6.6.75-current-sunxi` to `6.12.23-current-sunxi`

EvilOlaf, 2025-05-21:
Downgrading u-boot package only replaces binaries available for flashing. Though flashing does not happen automatically since it is rarely necessary to actually do that.

igorpecovnik, 2025-06-29:
Possible solution: https://forum.armbian.com/topic/22810-trying-pcduino2-on-a-newer-kernel/

sp2ong, 2025-08-15:
Sorry, but it looks like used U-Boot 2025.07

[image]

igorpecovnik, 2025-08-15:
This is u-boot add-on which is packaged, but it updating goes via armbian-config - update bootloader.

igorpecovnik, 2025-08-15:
You can generate u-boot from that PR, install u-boot package manually and update via armbian-config ... but its not needed as it was verified by @d93921012

leggewie, 2025-08-15:
closed via #8503

sp2ong, 2025-08-16:
I try get an update and upgrade on my OZPI v1 to check reboot but the whole day is 

Err:2 http://apt.armbian.com bullseye/main armhf armbian-zsh all 25.8.1
  File has unexpected size (12001044 != 11982476). Mirror sync in progress? [IP: 46.102.157.202 80]
Get:3 http://armbian.systemonachip.net/apt bullseye/main armhf linux-dtb-current-sunxi armhf 25.8.1 [211 kB]
Err:3 http://armbian.systemonachip.net/apt bullseye/main armhf linux-dtb-current-sunxi armhf 25.8.1
  File has unexpected size (210992 != 210996). Mirror sync in progress? [IP: 46.102.157.202 80]
  Hashes of expected file:
   - SHA512:857b35e08cada5a1d950ec0b9a32e6d1a11eca1b6b1d6a0d107556a8f2b54e721f031c21c8d36335694a0c72e659615af9ff7dc56a8803c7a809937a09396f8d
   - SHA256:06577b012018e5790ffa75d7ac4f224bee8afecdec92ec7d59ea60e201c31a1c
   - SHA1:a1d3cbd97248985b8a6d06a4f5f109a0759c66ef [weak]
   - MD5Sum:4e12ecbeae20246b2ecb1714b4bb41b8 [weak]
   - Filesize:210996 [weak]

sp2ong, 2025-08-16:
OK,

On another OZPI v1 with bookworm problem, connect to fi.mirror.armbian.de so I suppose tomorrow all will be back to normal :-)

Hit:1 http://deb.debian.org/debian bookworm InRelease
Get:2 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:3 http://security.debian.org bookworm-security InRelease
Get:4 http://deb.debian.org/debian bookworm-backports InRelease [59.4 kB]
Hit:6 https://github.armbian.com/configng stable InRelease
Ign:5 http://fi.mirror.armbian.de/beta bookworm InRelease                                                              
Ign:5 http://fi.mirror.armbian.de/beta bookworm InRelease
Ign:5 http://fi.mirror.armbian.de/beta bookworm InRelease
Err:5 http://fi.mirror.armbian.de/beta bookworm InRelease
  Could not connect to fi.mirror.armbian.de:80 (65.21.120.247), connection timed out
Fetched 115 kB in 38s (3,015 B/s)
Reading package lists... Done
W: Failed to fetch http://beta.armbian.com/dists/bookworm/InRelease  Could not connect to fi.mirror.armbian.de:80 (65.21.120.247), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.

EvilOlaf, 2025-08-16:
> fi.mirror.armbian.de

Just checked the machine, runs normal. Slightly slower than usual but that's it. https://status.armbian.de/endpoints/2-mirror_fi-mirror

sp2ong, 2025-08-16:
At current

The following packages will be upgraded:
  armbian-bsp-cli-orangepizero-current armbian-firmware base-files linux-dtb-current-sunxi linux-image-current-sunxi
  linux-u-boot-orangepizero-current
6 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 139 MB of archives.
After this operation, 18.4 kB disk space will be freed.
Do you want to continue? [Y/n] 
Ign:1 http://beta.armbian.com bookworm/bookworm-utils armhf base-files armhf 25.11.0-trunk.31-12.4+deb12u11-bookworm
Ign:2 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-bsp-cli-orangepizero-current armhf 25.11.0-trunk.31
Ign:3 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-firmware all 25.11.0-trunk.31
Ign:4 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-dtb-current-sunxi armhf 25.11.0-trunk.31
Ign:5 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-image-current-sunxi armhf 25.11.0-trunk.31
Ign:6 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-u-boot-orangepizero-current armhf 25.11.0-trunk.31
Ign:1 http://beta.armbian.com bookworm/bookworm-utils armhf base-files armhf 25.11.0-trunk.31-12.4+deb12u11-bookworm
Ign:2 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-bsp-cli-orangepizero-current armhf 25.11.0-trunk.31
Ign:3 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-firmware all 25.11.0-trunk.31
Ign:4 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-dtb-current-sunxi armhf 25.11.0-trunk.31
Ign:5 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-image-current-sunxi armhf 25.11.0-trunk.31
Ign:6 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-u-boot-orangepizero-current armhf 25.11.0-trunk.31
Ign:1 http://beta.armbian.com bookworm/bookworm-utils armhf base-files armhf 25.11.0-trunk.31-12.4+deb12u11-bookworm
Ign:2 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-bsp-cli-orangepizero-current armhf 25.11.0-trunk.31
Ign:3 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-firmware all 25.11.0-trunk.31
Ign:4 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-dtb-current-sunxi armhf 25.11.0-trunk.31
Ign:5 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-image-current-sunxi armhf 25.11.0-trunk.31
Ign:6 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-u-boot-orangepizero-current armhf 25.11.0-trunk.31
Err:1 http://beta.armbian.com bookworm/bookworm-utils armhf base-files armhf 25.11.0-trunk.31-12.4+deb12u11-bookworm
  Could not connect to fi.mirror.armbian.de:80 (65.21.120.247), connection timed out
Err:2 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-bsp-cli-orangepizero-current armhf 25.11.0-trunk.31
  Unable to connect to fi.mirror.armbian.de:http:
Ign:3 http://fi.mirror.armbian.de/beta bookworm/main armhf armbian-firmware all 25.11.0-trunk.31
Err:3 http://beta.armbian.com bookworm/main armhf armbian-firmware all 25.11.0-trunk.31
  Unable to connect to fi.mirror.armbian.de:http:
Err:4 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-dtb-current-sunxi armhf 25.11.0-trunk.31
  Unable to connect to fi.mirror.armbian.de:http:
Err:5 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-image-current-sunxi armhf 25.11.0-trunk.31
  Unable to connect to fi.mirror.armbian.de:http:
Err:6 http://fi.mirror.armbian.de/beta bookworm/main armhf linux-u-boot-orangepizero-current armhf 25.11.0-trunk.31
  Unable to connect to fi.mirror.armbian.de:http:
E: Failed to fetch http://fi.mirror.armbian.de/beta/pool/bookworm-utils/b/base-files/base-files_25.11.0-trunk.31-12.4+deb12u11-bookworm_armhf__1-bookworm-1armbian1-Baa17-U12.4--deb12u11-R55fa.deb  Could not connect to fi.mirror.armbian.de:80 (65.21.120.247), connection timed out
E: Failed to fetch http://fi.mirror.armbian.de/beta/pool/main/a/armbian-bsp-cli-orangepizero-current/armbian-bsp-cli-orangepizero-current_25.11.0-trunk.31_armhf__1-PCee86-V6766-H1893-B1e5d-Rf36a.deb  Unable to connect to fi.mirror.armbian.de:http:
E: Failed to fetch http://beta.armbian.com/pool/main/a/armbian-firmware/armbian-firmware_25.11.0-trunk.31_all__1-SA105f-B6c7f-R448a.deb  Unable to connect to fi.mirror.armbian.de:http:
E: Failed to fetch http://fi.mirror.armbian.de/beta/pool/main/l/linux-dtb-current-sunxi/linux-dtb-current-sunxi_25.11.0-trunk.31_armhf__6.12.35-S783c-D0b5d-P0ef8-C34a1Hf8af-HK01ba-V014b-Bd200-R448a.deb  Unable to connect to fi.mirror.armbian.de:http:
E: Failed to fetch http://fi.mirror.armbian.de/beta/pool/main/l/linux-6.12.35/linux-image-current-sunxi_25.11.0-trunk.31_armhf__6.12.35-S783c-D0b5d-P0ef8-C34a1Hf8af-HK01ba-V014b-Bd200-R448a.deb  Unable to connect to fi.mirror.armbian.de:http:
E: Failed to fetch http://fi.mirror.armbian.de/beta/pool/main/l/linux-u-boot-orangepizero-current/linux-u-boot-orangepizero-current_25.11.0-trunk.31_armhf__2024.01-S866c-P7738-Ha5c2-Vd711-Bb703-R448a.deb  Unable to connect to fi.mirror.armbian.de:http:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?