armbian/build #8227: rock-5b-plus u-boot build error

CodeChenL, 2025-05-27:
error log: https://github.com/armbian/os/actions/runs/15214928430/job/42800625037#step:8:982

As a result, the `u-boot.itb` in the built deb package does not contain the device tree and other content:

```
[chen@ChenArchLinux 15:07]
[0 /tmp/ark-mOgieT/usr/lib/linux-u-boot-vendor-rock-5b-plus]$ binwalk u-boot.itb 

                                             /tmp/ark-mOgieT/usr/lib/linux-u-boot-vendor-rock-5b-plus/u-boot.itb
-------------------------------------------------------------------------------------------------------------------------------------------------------------
DECIMAL                            HEXADECIMAL                        DESCRIPTION
-------------------------------------------------------------------------------------------------------------------------------------------------------------
0                                  0x0                                Device tree blob (DTB), version: 17, CPU ID: 0, total size: 1971 bytes
793612                             0xC1C0C                            CRC32 polynomial table, little endian
-------------------------------------------------------------------------------------------------------------------------------------------------------------

Analyzed 1 file for 85 file signatures (187 magic patterns) in 15.0 milliseconds
```
Trying to rebuild the uboot deb package using `armbian/build` does not cause this problem

igorpecovnik, 2025-05-27:
It is probably related to compiler change - we switched to Ubuntu Noble recently.

https://github.com/armbian/build/pull/8220

HeyMeco, 2025-06-16:
For comparison these are the build logs for u-boot on:
Jammy Log: https://paste.armbian.com/icunejacix
Noble Log: https://paste.armbian.com/awadewuzif

While both build without errors only Jammy produces a working spi image. (EMMC/SD-Card boot not affected)

amazingfate, 2025-06-16:
While I can build working uboot on arm64 noble without docker. But the uboot built by workflow can't boot.

HeyMeco, 2025-06-16:
@amazingfate 
> While I can build working uboot on arm64 noble without docker. But the uboot built by workflow can't boot.

The logs were from my arm64 debian host with docker.
Both variants boot via EMMC but with the rkspi img built from Noble-Docker the status LED stays green while Jammy-Docker turns blue and boots the test NVMe I have

igorpecovnik, 2025-06-17:
Beware that we are overriding BASE_DOCKER_IMAGE in configs:
https://github.com/armbian/os/blob/main/userpatches/config-armbian-community.conf#L21
https://github.com/armbian/os/blob/main/userpatches/config-armbian-images.conf#L22-L23
As otherwise there are still troubles.

CodeChenL, 2025-06-18:
> Now [the latest image](https://github.com/armbian/os/releases/download/25.8.0-trunk.232/Armbian_25.8.0-trunk.232_Rock-5b-plus_trixie_vendor_6.1.115_minimal.img.xz) built from workflow shows correct binwalk output:现在从工作流程构建的最新镜像显示了正确的 binwalk 输出：
> 
> ```
> $ binwalk /mnt/usr/lib/linux-u-boot-vendor-rock-5b-plus/u-boot.itb 
> 
> DECIMAL       HEXADECIMAL     DESCRIPTION
> --------------------------------------------------------------------------------
> 0             0x0             Flattened device tree, size: 1971 bytes, version: 17
> 793620        0xC1C14         CRC32 polynomial table, little endian
> 831429        0xCAFC5         Intel x86 or x64 microcode, sig 0xe6001916, pf_mask 0xc101c1fe, 1C1C-13-10, rev 0x1c0d0a1f, size 255
> 904092        0xDCB9C         Android bootimg, kernel size: 1684095488 bytes, kernel addr: 0x646E6120, ramdisk size: 1684631410 bytes, ramdisk addr: 0x616D6920, product name: "der"
> ```
> 
> [@CodeChenL](https://github.com/CodeChenL) is this image bootable?这张镜像可启动吗？

Thank you for the reminder, I will test it when I have time

rpardini, 2025-11-12:
This has come up again, due to all/most armbian/os builds having been changed to ubuntu:jammy (while armbian/build defaults to noble for the Docker scenario). 

We should introduce binwalk's of all the produced files in the u-boot deb in the build logs, so we could more easily compare binwalks between what works and what doesn't, and then maybe we can divine what linker/compiler flag (or whatever...) is needed to make it work again on modern build hosts.

@HeyMeco reports the 3576 + UFS case only works on Jammy, not on Noble, and not on Trixie. I wonder what Bookworm does, etc.

igorpecovnik, 2026-08-07:
Yes, closing.

closed 2026-08-07 by igorpecovnik